ros/ros_comm #1535: "genmsg.msg_loader.MsgNotFound: Cannot locate message [Vector3]"

jingedawang, 2018-11-20:
Error when I run `rostopic echo -b cmd_vel_bug.bag /cmd_vel`. Error message shows following:
```
Traceback (most recent call last):
  File "/opt/ros/kinetic/bin/rostopic", line 35, in <module>
    rostopic.rostopicmain()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostopic/__init__.py", line 2093, in rostopicmain
    _rostopic_cmd_echo(argv)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostopic/__init__.py", line 1392, in _rostopic_cmd_echo
    _rostopic_echo(topic, callback_echo, bag_file=options.bag)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostopic/__init__.py", line 1013, in _rostopic_echo
    _rostopic_echo_bag(callback_echo, bag_file)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostopic/__init__.py", line 991, in _rostopic_echo_bag
    for t, msg, timestamp in b.read_messages():
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 2337, in read_messages
    yield self.seek_and_read_message_data_record((entry.chunk_pos, entry.offset), raw)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 2475, in seek_and_read_message_data_record
    msg_type = _get_message_type(connection_info)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 1571, in _get_message_type
    message_type = genpy.dynamic.generate_dynamic(info.datatype, info.msg_def)[info.datatype]
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/dynamic.py", line 155, in generate_dynamic
    for l in msg_generator(msg_context, spec, search_path):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/generator.py", line 742, in msg_generator
    genmsg.msg_loader.load_depends(msg_context, spec, search_path)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 355, in load_depends
    return load_msg_depends(msg_context, spec, msg_search_path)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 324, in load_msg_depends
    depspec = load_msg_by_type(msg_context, resolved_type, search_path)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 125, in load_msg_by_type
    file_path = get_msg_file(package_name, base_type, search_path)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 84, in get_msg_file
    % (base_type, package, search_path), base_type, package, search_path)
genmsg.msg_loader.MsgNotFound: Cannot locate message [Vector3]: unknown package [geometry_msgs] on search path [{}]
```
But it is ok when I run a different bag file `rostopic echo -b cmd_vel.bag /cmd_vel`.
I thought there may be anything wrong with cmd_vel_bug.bag, but `rosbag play cmd_vel_bug.bag` still work well.

Meanwhile, all the rosbag API which want to extract messages in /cmd_vel topic from cmd_vel_bug.bag also failed. For example, following code throws the same error as before:
```python
import rosbag

with rosbag.Bag('output.bag', 'w') as outbag:
    for topic, msg, t in rosbag.Bag('cmd_vel_bug.bag').read_messages(["/cmd_vel"]):
        # This also replaces tf timestamps under the assumption 
        # that all transforms in the message share the same timestamp
        outbag.write(topic, msg, t)
```

The two bags have been uploaded to Google Drive: [cmd_vel.bag](https://drive.google.com/file/d/1OjpwRX_JZ4jScYvlLxmlZ9E74kbgYbvb/view?usp=sharing), [cmd_vel_bug.bag](https://drive.google.com/file/d/12qyxuJxTGpPAGfJSvR7Z9Yd0ORWe7CRo/view?usp=sharing).

cwecht, 2019-05-14:
A rosbag contains embedded message definitions. The message definition of ``geometry_msgs/Twist`` in ``cmd_vel_bug.bag`` is this:
~~~
# This expresses velocity in free space broken into its linear and angular parts.
Vector3  linear
Vector3  angular
~~~
The message definition of ``geometry_msgs/Twist`` in ``cmd_vel.bag`` is this:
~~~
# This expresses velocity in free space broken into its linear and angular parts.
Vector3  linear
Vector3  angular

================================================================================
MSG: geometry_msgs/Vector3
# This represents a vector in free space. 
# It is only meant to represent a direction. Therefore, it does not
# make sense to apply a translation to it (e.g., when applying a 
# generic rigid transformation to a Vector3, tf2 will only apply the
# rotation). If you want your data to be translatable too, use the
# geometry_msgs/Point message instead.

float64 x
float64 y
float64 z
~~~

This is the problem, because this information is used to identify the type for Vector3. 

@jingedawang do you know, with which version of ``rosbag`` ``cmd_vel_bag.bag`` has been created? This information might help fixing the issue.

jingedawang, 2019-05-16:
Thank you very much. 
It has been a long time, there is no extra information I can provide. Could you tell me how to display the definitions above?

cwecht, 2019-05-16:
This script will give you the embedded message definition:
~~~
import rosbag

c = next(rosbag.Bag('cmd_vel.bag')._get_connections(["/cmd_vel"]))
print(sc.msg_def)
~~~

closed 2019-05-16 by jingedawang