ros/ros_comm #1538: Using .test file for multiple tests

aswinthomas, 2018-11-20:
Using ARG input we can set multiple params for the .test file, but can we use the same test file with different parameters to run the same tests with different inputs? How do we do this? 
Put another way, is there a way we can run parametrised rostests like in gtests

dirk-thomas, 2018-11-20:
I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2018-11-20 by dirk-thomas

aswinthomas, 2018-11-20:
Noted. Posted [here](https://answers.ros.org/question/308987/how-to-run-multiple-rostests-with-one-test-file-and-one-gtest-test-case/)