ros/ros_comm #1542: Is Timer.hasPending broken?

kunaltyagi, 2018-11-22:
Going by the docs, `hasPending` should return true if there's a callback which would be called in future.

```
Returns whether or not the timer has any pending events to call
```
My reading of `event to call` is a callback,.

My expectations:
* For invalid, stopped and unstarted timers return false
* For oneshot timers, it returns true before the callback is called and false afterwards
* For other timers, returns true all the time

Reality:
* Always false in my test scenarios

Either the docs are confusing or the functionality is broken

cwecht, 2019-04-23:
After having a look at the code, I think, that ``hasPending`` tells the caller, whether the timer has any pending callback in the callback queue, which is the same used for e.g. the message callbacks. Timer-callbacks are added to this queue at the time the callback is meant to be called (this description is not accurate but sufficient for this purpose). 

Let's have a look at the following example:

~~~~
#include <ros/ros.h>

void timerCallback(const ros::TimerEvent& event) {
  ros::Duration(1.0).sleep();
}
  
int main(int argc, char *argv[]) {
  ros::init(argc, argv, "test_node");
  ros::NodeHandle nh;
  ros::Timer timer = nh.createTimer(ros::Duration(1.0/5.0), timerCallback);
    
  while (ros::ok())  {
    ros::spinOnce();
    std::cout << "hasPending: " << (timer.hasPending() ? "true" : "false") << std::endl;
  }
}
~~~~

In this example, there are more callbacks added to the queue than a single thread can process, because the timers period is shorter than the duration of a callback. The timer has now callbacks, which are waiting in the callback queue pending, so the output is ``hasPending: true``.

kunaltyagi, 2019-04-24:
Thanks a lot @cwecht.
If I understand correctly, `hasPending` isn't a function to check the timer, but instead its callback queue.

Is there some way I can access the pending status for future callbacks (true if a callback would be called, false otherwise) instead of the callback queue?

fujitatomoya, 2019-04-24:
@kunaltyagi 

already mentioned by @cwecht , but TimerManager thread func just appends the callback into the callback queue as scheduled, hasPending is to check the waiting_callbacks.

what do you mean by "pending status"? can you be more specific?

kunaltyagi, 2019-04-24:
Maybe "pending status" isn't the right word to use. Perhaps "is_alive" or "is_active"  are better terms to use.

I want to check if the timer should be deleted or should be kept by checking if it'll receive callbacks in the future. The timer itself is kept so that it's period can be adjusted. Sadly, I get a lot of timers in my memory and I want to delete the pointers of the timers which I don't really care about. My deleting the pointers reduces the reference count and can trigger any cleanup (if required) in the artefacts created by nodehandle.

To find this out, do I need to perform external bookkeeping? I think the timer class is internally 
performing the requisite steps already to schedule callbacks somehow.

fujitatomoya, 2019-04-24:
@kunaltyagi 

at least, you can see if the callbacks are pending(will be fired) or not via hasPending, right?
sorry, not very much sure what you are trying to.
the timer just works with the user requests. so why dont you just do reconsider or cleanup your request for time?

kunaltyagi, 2019-04-24:
@fujitatomoya I have to keep a timer around IFF it will call the callback sometime in future. My node provides some functions which can be called with a delay or periodically. Please check the second snippet for what I envision the usecase to be.

```cpp
bool create_callback(request, response) {
    // simplified form
    auto iter = my_map.find(request.name);
    if (iter != my_map.end()) { return false; }
    my_map[request.name] = nh.createTimer(request.timeout,
                                          create_lambda(request.callback),
                                          request.oneshot);
    return true;
}

bool reset_timeout_callback(request, response) {
    // simplified form
    auto iter = my_map.find(request.name);
    if (iter == my_map.end()) { return false; }
    *iter.setPeriod(request.timeout, request.reset);
    return true;
}
```

Trim down in loop
```cpp
while(ros::ok()) {
    for (auto kv = my_map.begin(); kv != my_map.end()) {
        // The following  check is not possible using kv->second.hasPending()
        // for reasons explained by @cwecht
        if (kv->second.is_active()) { kv++; }
        else { my_map.erase(kv++); }
    }
    // we only have timers which are required in the future.
    // All timers created by ros::Timer() are removed
    // All one-shot timers which have called once are removed
    // All other timers (non-oneshot and uncalled one-shot) are NOT removed

    // Why check? The code will work well if there's no check. O(log_2 N) is sufficient for scaling
    // upto thousands of elements.
    // Ans: Some more processing is performed based on which lambdas are to be called in future
}
```

closed 2025-05-31 by sloretz