ros/ros_comm #1552: topic_tools symbol lookup error

BoukeKromTNO, 2018-11-30:
ROS kinetic release on Ubuntu 16.04
```
apt list ros-kinetic-topic-tools
>> ros-kinetic-topic-tools/xenial,now 1.12.14-0xenial-20180824-102957-0800 amd64 [installed]
```
I get a symbol lookup error on running `relay` or `mux`, haven't tested others.
```
rosrun topic_tools relay /topic1 /topic2
>> /opt/ros/kinetic/lib/topic_tools/relay: symbol lookup error: /opt/ros/kinetic/lib/topic_tools/relay: undefined symbol: _ZN11topic_tools11getBaseNameERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERS5_

rosrun topic_tools mux topic1 topic2
/opt/ros/kinetic/lib/topic_tools/mux: symbol lookup error: /opt/ros/kinetic/lib/topic_tools/mux: undefined symbol: _ZN11topic_tools11getBaseNameERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERS5_
```
Without arguments runs as expected:
```
rosrun topic_tools relay
>> usage: relay IN_TOPIC [OUT_TOPIC]
```

Tested on three different systems with similar installs. I'd be happy to provide more info to track down the cause of this issue.

cwecht, 2019-04-17:
I wasn't able to reproduce this. with the ``ubuntu:xenial`` docker image. @BoukeKromTNO is this still an issue?

BoukeKromTNO, 2019-04-18:
Indeed, has fixed itself by now!

closed 2019-04-18 by BoukeKromTNO