ros/ros_comm #1556: reused rosbag::Bag instance crashes

denwi248, 2018-12-05:
I'm on Ubuntu 16.04 with ros kinetic 1.12.14.

A minimal reproducible example is:

    #include <rosbag/bag.h>
    #include <rosbag/view.h>
    #include <std_msgs/Int32.h>

    rosbag::Bag bag;

    // generate first bag file
    bag.open("test.bag", rosbag::bagmode::Write);

    std_msgs::Int32 msg;

    for ( unsigned int i = 0; i < 10; ++i ) {
        msg.data = i;
        bag.write("numbers", ros::Time::now(), msg);
    }

    bag.close();

    // generate second bag file
    bag.open("test2.bag", rosbag::bagmode::Write);

    for ( unsigned int i = 10; i < 20; ++i ) {
        msg.data = i;
        bag.write("numbers2", ros::Time::now(), msg);
    }

    bag.close();

    // read in first bag file
    bag.open("test.bag", rosbag::bagmode::Read);

    for ( rosbag::MessageInstance const m : rosbag::View(bag) ) {
        std_msgs::Int32::ConstPtr i = m.instantiate<std_msgs::Int32>();
        if ( i != nullptr )
          std::cout << "bag1: " << i->data << std::endl;
    }

    bag.close();

    // read in second bag file
    bag.open("test2.bag", rosbag::bagmode::Read);

    for ( rosbag::MessageInstance const m : rosbag::View(bag) ) {
        std_msgs::Int32::ConstPtr i = m.instantiate<std_msgs::Int32>();
        if ( i != nullptr )
          std::cout << "bag2: " << i->data << std::endl;
    }

    bag.close();

I create two bag files and want to read them in later. All with the same rosbag::Bag instance. This works until I open the second bag file and crashes with the error message:

> Error:   Received an invalid TCPROS header.  Each element must be prepended by a 4-byte length. 
at line 88 in /tmp/binarydeb/ros-kinetic-cpp-common-0.6.11/src/header.cpp                                         
terminate called after throwing an instance of 'rosbag::BagFormatException'                                                              
  what():  Error parsing header

I found a workaround by using a new instance of rosbag::Bag for the second rosbag file. This is working as expected.

Strange side note:
when I change the topic name in the second bag file from _numbers2_ to _numbers_, then it does not crash but it is printing two times numbers from 0-9 instead 0-9 and 10-19. So it looks like it has opened the first bag file again.

denwi248, 2018-12-05:
Thanks for the hint. Will that fix be ported to kinetic, too?

racko, 2018-12-05:
The PR is referenced by [changes between 1.12.7 and 1.13.4 for backporting](https://github.com/ros/ros_comm/pull/1205). There #1192 is listed as "Not backported":
> patch would require other changes to be backported or patch to be rewritten