ros/ros_comm #1558: roscpp spin() vs Timer

msadowski, 2018-12-06:
Hi,

I've been testing a lot moving the system clock backward and forward in time, while having a timer callback running in the node.

The code comes down to the following lines:
```
  timer_ = nh.createTimer(ros::Duration(1.0/50.0), &VescDriver::timerCallback, this);
  ros::Rate loop_rate(50);
  while (ros::ok())
    // ros::spin();  
  {
    loop_rate.sleep();
    ros::spinOnce();
  }
...
void VescDriver::timerCallback(const ros::TimerEvent& event)
{
  ROS_INFO("TIMER CALLBACK"); //TODO: REMOVE (TESTING TIMER RESET)
}
```

I noticed that if I have a `ros::spin()` enabled and move the time back using a command `sudo date +%T -s "10:23:00" ` the timerCallback stops firing.

If on the other hand I execute the busy waiting loop manually and call `ros::spinOnce()` after each pass of the loop the event is firing as expected when I move the time (both forward and backward).

Is this an expected behaviour? If yes I think it would be worth documenting this.

cwecht, 2019-03-29:
As far as I can tell, the reason for this issue is, that ros::Timer uses ``boost::condition_variable`` for timed waiting (basically interruptable sleeping), which is based on ``pthread_cond_timedwait``, which wait until a fix time point. Therefore, if time jumps back, the timer waits until the time point is reached.

Currently I see two ways to fix this:
1. Do not rely on ``boost::condition_variable``; use a loop with ``ros::Duration()::sleep()``.
2.  Introduce another thread in ``TimerManager``, which checks, if time has jumped back.

cwecht, 2019-03-29:
Or an implementation derived from ``SteadyTimer`` could be used.

cwecht, 2019-04-05:
#1651 should fix this.

fujitatomoya, 2019-04-11:
as described above, this is known issue and you can use SteadyTimer to fix this.

msadowski, 2019-04-11:
@cwecht @fujitatomoya thanks for the heads up! Looking forward to see the mentioned PR merged!

closed 2025-05-31 by sloretz