ros/ros_comm #1574: Conformance with C++ standard timers

kunaltyagi, 2019-01-09:
Sometimes, I find the need to switch from system clock to ROS clock and it's not straightforward.

I read the C++ standard (a bit lazily) and found that the following code allows me to switch from `std::chrono::steady_clock` to `ros::my_clock`.

I don't know the following details
* type of clock used by `ros::Time::now()`
* where would this code go
* correctness of period as `std::ratio<1,1>`

```
namespace ros {
struct my_clock : public ros::Time
{
    using rep = double;
    using period = std::ratio<1, 1>;
    using duration = ros::Duration;
    using time_point = ros::Time;
    constexpr static bool is_steady(void) noexcept { return false; }
    static time_point now(void) noexcept(noexcept(ros::Time::now()))
    {
        return ros::Time::now();
    }
};
} // namespace clock
```

cwecht, 2019-02-27:
As long as `ros::WallTime::now()` uses a time-source, which is not `std::chrono`-based, your question can not be anwsered reliably. For example Linux provides about three different clocks. As you can see [here](https://github.com/ros/roscpp_core/blob/d95ac897dd759188182395b1cbcf40f37207dd3e/rostime/src/time.cpp#L107) supports two different clock sources. libstdc++ seems to use the same ([have a look here](https://github.com/gcc-mirror/gcc/blob/4bc701d37fb90229e7337bfd8f510bb74fb1aadf/libstdc%2B%2B-v3/src/c%2B%2B11/chrono.cc)), but I think, that we should not rely on that. libc++ (LLVM) might do it differently and on Windows/Android this is a whole different story.

I'd discourage the use of differenct clouck sources. Use either `ros::Time::now()` or `std::chrono::steady_clock` but not both. Since ROS-components usually use `ros::Time::now()`/`ros::WallTime::now()`, I'd encourage  you to use `ros::Time::now()`.

If you need any other chrono-functionality, you can write conversion functions in a straight forward manner.

In the future, we could think about using `std::chrono` as a portable backend of `ros::Time`, but this would be be a rather extensive change.

flixr, 2019-02-27:
If you want to use the steady clock there is `ros::SteadyTime`.

kunaltyagi, 2019-03-08:
I have some code which is non-ROS and am porting it to work with ROS. For that, both clocks require same interface to access the time, and since `boost` and `c++11` have the same structure, I found it easier to create a Clock wrapper for ROS Time. I'm proposing to create the same interface in ROS as in Boost and C++ standard for chrono utilities because it allows people to use code with similar semantics.

Eg: `ROS::steady_clock` vs `boost::chrono::steady_clock` vs `std::chrono::steady_clock` only require changing the class (well namespace, but templating over class is easier)

Currently, ROS has `Time` and not `Clocks`. I hope to constrain the discussion solely to adding clocks in ROS so that people (hopefully others apart from just me :wink: ) can use clocks without worrying about syntax error in moving from one library to another. Having a standard back-end for ROS is an implementation detail which is a different decision.

PS: If you didn't know about Boost Thread Clock and are debugging threading issues, look into it :smile:

closed 2025-05-31 by sloretz