ros/ros_comm #1575: rospy client does not cache parameters

NickZ, 2019-01-10:
rospy client does not cache parameters despite having a parameter cache, and there is no option to enable it:
https://github.com/ros/ros_comm/blob/melodic-devel/clients/rospy/src/rospy/msproxy.py#L119

This is in contrast to roscpp which has the option to cache parameters, and it is on by default.

This is easily observed by running the "talker" tutorial for both roscpp and rospy and monitoring communications on port 11311 with wireshark. The rospy version will continuously send a `getParam` request for the parameter `/rosout_disable_topics_generation` without subscribing, the roscpp will only send one and will subscribe to the parameter.

Since roscpp does cache parameters, it makes sense to enable this for rospy as well, unless I am missing something.

mgrrx, 2019-01-10:
Checkout https://github.com/ros/ros_comm/pull/1515

mgrrx, 2019-01-12:
Interesting, seems like https://github.com/ros/ros_comm/pull/1241 introduced this regression in melodic. I didn't notice this yet since at Magazino we are still on kinetic and this feature was not backported.

paulbovbel, 2019-01-31:
I think the more appropriately alarming title for this issue is 'rospy hits rosmaster on every log invocation'.

Thanks for the caching fix @mgrrx, we're going to run our robots on https://github.com/locusrobotics/ros_comm/pull/10 for a while as validation.

dirk-thomas, 2020-02-10:
This was resolved by #1515.

closed 2020-02-10 by dirk-thomas