ros/ros_comm #1598: Melodic source repository has got broken on Xenial

slavanap, 2019-01-29:
`wstool init` started to fail to retrieve several ros_comm packages several days ago for Melodic on Xenial

Working build log (wstool succeeds)
https://gitlab.com/slavanap/ros-build/-/jobs/152120802

Broken build log for the same bash commands (wstool fails)
https://gitlab.com/slavanap/ros-build/-/jobs/152547627

In case you need any additional info or samples, please inform me.

Thanks,
Vyacheslav

dirk-thomas, 2019-01-29:
The referenced are actually still there (checked by cloning the repo) and haven't changed since they were created. It currently looks like a problem with GitHub not being able to serve the requested resources.

Some more context:
* the repo in question: https://github.com/ros-gbp/ros_comm-release
* the tag expected to exist: `release/melodic/message_filters/1.14.3-0` (`git tag -l` in a local working copy does show the tag exists)
* :heavy_exclamation_mark: the "switch-to-tags" drop down list doesn't contain the tag
* :heavy_check_mark: the releases page does show the tag: https://github.com/ros-gbp/ros_comm-release/releases?after=release%2Fmelodic%2Frosmaster%2F1.14.3-0
* :heavy_exclamation_mark: the deep link doesn't work: https://github.com/ros-gbp/ros_comm-release/archive/release/melodic/message_filters/1.14.3-0.tar.gz
https://github.com/ros-gbp/ros_comm-release/archive/release/melodic/message_filters/1.14.3-0.tar.gz

slavanap, 2019-01-29:
The build fails again. I've just noticed tgat versions 1.14.3 & 1.14.4 are mixed in the log. May it be some dependency resolution algorithm that breaks it?

slavanap, 2019-02-01:
@dirk-thomas I've located the bug. The error raises here, after the download: https://github.com/vcstools/vcstools/blob/master/src/vcstools/tar.py#L125-L126

Now I don't think its related to GitHub CDN, but relates to untar algorithm or something.

And ofc, this issue is not a question.

slavanap, 2019-02-01:
For instance
```Error processing 'ros_comm/rosbag_storage' : [ros_comm/rosbag_storage] Checkout of https://github.com/ros-gbp/ros_comm-release/archive/release/melodic/rosbag_storage/1.14.3-0.tar.gz version ros_comm-release-release-melodic-rosbag_storage-1.14.3-0 into /src/base/rosbase/src/ros_comm/rosbag_storage failed.
```
If you download and unpack the tar, you'll see `ros_comm-release-release-melodic-rosbag_storage` folder, that doesn't match `ros_comm-release-release-melodic-rosbag_storage-1.14.3-0` version.
The question is from where this version comes?

dirk-thomas, 2019-02-01:
> And ofc, this issue is not a question.

The "updated" definition of the `questions` label on GitHub is:

> > Further information is requested

While this is an older repo and we haven't added the descriptions to labels we treat them in the same way as newer repos. So basically we need further information before we can act on it.

dirk-thomas, 2019-02-01:
> I've located the bug. The error raises here, after the download: https://github.com/vcstools/vcstools/blob/master/src/vcstools/tar.py#L125-L126

I am not familiar with all the custom logic applied in `vcstools`. I only use `vcstool` (without the trailing `s`) - maybe you can give that one a try and see if it makes it work for your use case.

Otherwise I would suggest to open a ticket in the `vcstools` repo.

mryellow, 2019-02-02:
@slavanap Confirmed on Debian Stretch that removing numbers from `version` works around whatever has been modified in the release process.

```
- tar :
    local-name: ros_comm/message_filters
    uri: https://github.com/ros-gbp/ros_comm-release/archive/release/melodic/message_filters/1.14.3-0.tar.gz
    version: ros_comm-release-release-melodic-message_filters
```

```
- tar:
    local-name: ros_comm/message_filters
    uri: https://github.com/ros-gbp/ros_comm-release/archive/release/melodic/message_filters/1.14.3-0_28.tar.gz
    version: ros_comm-release-release-melodic-message_filters
```

slavanap, 2019-02-03:
@mryellow, yes. I'm testing similar workaround ATM
```
cat abc.rosinstall | perl -pe \
    's/(version:\s+ros_comm-release-release-melodic-)(message_filters|rosbag_storage|rosgraph|roslaunch|rosparam|rospy|rosservice|rostopic|roswtf|topic_tools|xmlrpcpp)-1.14.3-0/$1$2/g' \
    > abc.fixed.rosinstall
```

dirk-thomas, 2019-02-04:
Neither the existing tags in the release repo nor the release process has changed. You can confirm this by cloning the release repo and checking for the tag in question. The only problem is that GitHub doesn't serve the resources as it does for other tags / repos. We have reached out to GitHub last week and are waiting for a response (the first response only mentioned that they are looking into it. As mentioned [above](https://github.com/ros/ros_comm/issues/1598#issuecomment-458713201) a similar problem appeared on other repos as well.

dirk-thomas, 2019-02-08:
The GitHub support replied with a confirmation that they are able to reproduce the problem and are currently investigating...

slavanap, 2019-02-12:
FYI, some tar archives has been fixed. Now the workaround looks like this
```
cat abc.rosinstall | perl -pe \
    's/(version:\s+ros_comm-release-release-melodic-)(rosbag_storage|rosgraph)-1.14.3-0/$1$2/g' \
    > abc.fixed.rosinstall
```

leolchat, 2019-02-18:
It is still flickering back and forth with versions being added and removed. I have been trying to follow with those changes, but it is changing all the time...
@dirk-thomas do you have a link of the GitHub issue where we can track their progress / how to fix that?

dirk-thomas, 2019-02-19:
> do you have a link of the GitHub issue where we can track their progress / how to fix that?

I reported the problem through the GitHub feedback form. There is no public ticket tracking the problem afaik. I also waiting for further updates via email from GitHub support...

clalancette, 2019-02-19:
We're also tracking the issue here: https://github.com/vcstools/vcstools/issues/147

dirk-thomas, 2019-02-22:
> We're also tracking the issue here: [vcstools/vcstools#147](https://github.com/vcstools/vcstools/issues/147)

That issue seems to describe a different problem: different content of the archive vs. the archive not being fetchable.

slavanap, 2019-02-22:
@dirk-thomas  Archive is fetchable. Archive validation procedure leads to the error message above.

dirk-thomas, 2019-02-22:
> Archive is fetchable

For some time it wasn't fetchable (see https://github.com/ros/ros_comm/issues/1598#issuecomment-458711797). It is fetchable now and the only remaining issue with the GitHub interface is that it doesn't show the tag even though it exists in the repo (which has been reported to GitHub).

The problem due to the content change is tracked in the `vcstools` ticket and a straight forward workaround is to use `vcstool` instead.

Since these is nothing left to be done here I will go ahead and close this ticket.

closed 2019-02-22 by dirk-thomas