ros/ros_comm #1599: roscomm/ros_bag crypto implementation

CdeMills, 2019-01-30:
Hello,
after doing a manual install of ros (Melodic Morenia) on CentOS 7, I found a few issues in bag.py:
from Crypto import Random
from Crypto.Cipher import AES
Those routines are provided by pycrypto, which is unmaintained since 2013. Its replacement is cryptodome, which can be installed (https://pycryptodome.readthedocs.io/en/latest/src/installation.html) either as drop-in replacement, either as package "Cryptodome". The latter approach was used in Centos, so bag.py needs the following patch:
--- bag.py.orig	2019-01-30 13:46:25.700767679 +0100
+++ bag.py	2019-01-30 13:48:02.589286201 +0100
@@ -50,8 +50,13 @@
 import time
 import yaml
 
-from Crypto import Random
-from Crypto.Cipher import AES
+try:
+	from Crypto import Random
+	from Crypto.Cipher import AES
+except ImportError:
+	from Cryptodome import Random
+	from Cryptodome.Cipher import AES
+
 import gnupg

This is the "least surprise" approach: use pycrypto first, and cryptodome if not found.

Yet there are more issues lurking around. Around line 221:
        f.seek(chunk_data_pos)
         chunk = _read(f, chunk_size)
         # Encrypt chunk
        iv = Random.new().read(AES.block_size)
        f.seek(chunk_data_pos)
        f.write(iv)
pycryto uses its own entropy source, while cryptodome uses the correct os.urandom() approach. By comparing sources, its seems that pycrypto never called urandom and was subject to race conditions (CVE-2013-1445). The author of both modules is the same person, yet the evolution favored the dropping of the own implementation towards the use of the system-level random generator.

Would it be allowable to 
1) favor cryptodome over crypto
2) replace all lines like:
iv = Random.new().read(AES.block_size)
by direct call:
iv = os.urandom(AES.block_size)
to remove the extraneous interface layer of the Random class and ensure the right function is called?

Regards

Pascal

dirk-thomas, 2019-01-30:
This ticket replaces ros/ros#201.

Please create a pull request with the proposed patch.

dirk-thomas, 2019-01-31:
@CdeMills please do not open numerous similar PRs. Instead create one and discuss the possible deviations from the patch there.

dirk-thomas, 2019-08-07:
Closing due to no further reply / action.

PR #1609 (2019-08-14): Add pycryptodome as default on melodic
Fix #1599 

This patch will only work on Ubuntu 18.04 and up (Melodic and up). The package pycryptodome is not available on earlier version.
closed 2019-08-07 by dirk-thomas