ros/ros_comm #1600: rosbag: allow compilation on CentOS

CdeMills, 2019-01-30:
Hello,

CentOS 7 platform uses gcc-4.8.5; by default the c++ standard is 98. rosbag_storage and rosbag expect compilers using natively c++14. 

I produced a patch w.r.t files ros_comm/rosbag/CMakeLists.txt and 
ros_comm/rosbag_storage/CMakeLists.txt; unfortunately I can't add them to this issue. Basically it's about adding a few lines like 
target_compile_features(rosbag_storage PUBLIC cxx_std_11)
requesting that C++ standard is at least 11

Regards

Pascal

dirk-thomas, 2019-01-30:
Please provide the proposed patch in a pull request.

cwecht, 2019-02-22:
it seems, that `target_compile_features` requires at least cmake 3.1, so we must increase the minimum required cmake version. I would rather suggest to set `CMAKE_CXX_STANDARD` in the project (as it is suggested [here](https://crascit.com/2015/03/28/enabling-cxx11-in-cmake/)), which should be a simpler solution.

dirk-thomas, 2019-02-22:
> I would rather suggest to set `CMAKE_CXX_STANDARD` in the project (as it is suggested [here](https://crascit.com/2015/03/28/enabling-cxx11-in-cmake/)), which should be a simpler solution.

Please feel free provide a pull request.

I would expect that you will have the very same problem with a lot of ROS packages since ROS Melodic only targets platforms where the compilers default to C++14. So it might be more feasible to set the compiler option using an environment variable on CentOS 7 to make it work for all of them at once.

dirk-thomas, 2019-02-26:
Please see #1629 why this is being closed.

closed 2019-02-26 by dirk-thomas

cwecht, 2019-02-26:
I've just added a small note on this issue to ROS-Wiki's Troubleshooting page: http://wiki.ros.org/ROS/Troubleshooting#Failure_to_Compile