ros/ros_comm #1612: Issue with message_filters and PointCloud2 in python

pedghz, 2019-02-07:
I want to subscribe to pointcloud data and image data and sync them and then publish them again. I have written a short and simple script for this: 

    #!/usr/bin/python
    import rospy
    from sensor_msgs.msg import Image as SesnorImage
    from sensor_msgs.msg import CameraInfo
    from sensor_msgs.msg import PointCloud2
    import message_filters

    class synchronizer:
        def __init__(self):
            self.pub_Image = rospy.Publisher('image_raw_sync', SesnorImage, queue_size=1)
            self.pub_Cam_Info = rospy.Publisher('camera_info_sync', CameraInfo, queue_size=1)
            # self.pub_Lidar = rospy.Publisher('rslidar_points_sync', PointCloud2, queue_size=1)

            self.imageInput = message_filters.Subscriber('/image_raw', SesnorImage)
            self.cameraInfo = message_filters.Subscriber('/camera_info', CameraInfo)
            # self.lidar = message_filters.Subscriber('/rslidar_points', PointCloud2)

            self.ts = message_filters.TimeSynchronizer([self.imageInput,
                                                        self.cameraInfo,
                                                        # self.lidar
                                                        ], 10)
            self.ts.registerCallback(self.general_callback)

            self._image_raw = SesnorImage()
            self._camera_info = CameraInfo()
            self._rslidar_points = PointCloud2()

        def general_callback(self, image_raw,
                             camera_info,
                             # rslidar_points
                             ):
            self._image_raw = image_raw
            self._camera_info = camera_info
            # self._rslidar_points = rslidar_points

        def publisher(self):
            while True:
                self.pub_Image.publish(self._image_raw)
                self.pub_Cam_Info.publish(self._camera_info)
                # self.pub_Lidar.publish(self._rslidar_points)

    if __name__ == '__main__':
        rospy.init_node('synchronizer')
        synchronizer = synchronizer()
        synchronizer.publisher()
        rospy.spin()
    
But when I uncomment the commented lines, there would be no data in the published topics! so when I add the PointCloud2 sensor messages, everything gows bad and I dont know why this is happening. Why is this happening?

dirk-thomas, 2019-02-07:
I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2019-02-07 by dirk-thomas

pedghz, 2019-02-11:
Sure, here is the link. [https://answers.ros.org/question/315016/issue-with-message_filters-and-pointcloud2-in-python/](url)