ros/ros_comm #1613: Not able to open bag from file-like object

sripathivenky, 2019-02-15:
rosbag/bag.py:1112

 def _is_file(self, f):
        try:
            return isinstance(f, file)  # Python 2
        except NameError:
            import io
            return isinstance(f, io.IOBase)  # Python 3...this will return false in Python 2 always

line 1114 checks for instance of file in case of Python2.7 this causes issue if sile-like object was passed to Bag class.

line 119 of parameter definition says
"@param f: filename of bag to open or a stream to read from"

Cause of hard checking with type file in python 2.7 file-like objects can't be passed.

sripathivenky, 2019-02-15:
I would be happy to push the fix.

dirk-thomas, 2019-08-07:
Please create a pull request with your proposed fix then.

closed 2025-05-31 by sloretz