ros/ros_comm #1621: Topic Statistics rate under 0.016 for some connections

wentz89, 2019-02-20:
I have a plot-tool to visualize the statistics datas from rosbags and it seems that for some topic-connections there are only 7 messages over 420 sec on the /statistics topic, but with a pretty large amount of evaluated datas. I know that the statistics calculation uses a dynamic window if to less datas are available. This happened with more than one Topic. 

[image]

In the picture you can see the data plots i have taken.  In my opinion, however, this behavior is unintentional and unwanted. Thats why i open this issue.

wentz89, 2019-02-20:
Note: The /localization/odometry Topic has 2 other connections. These plots show a similar behavior, with a few more data-points (but not many).

dirk-thomas, 2019-02-20:
From the description of this ticket it is unclear what exactly your problem is, there is not enough information about what you are doing in order to reproduce what you are seeing and it is unclear if this is actually a bug in the sources of this repo or just related to what / how you are doing.

Therefore I would suggest to ask you question on [answers.ros.org](http://answers.ros.org/) instead. If you have asked your question there please add a link to the question to this ticket so that future readers can find the related question.

If that lead to enough evidence that there is actually a bug please feel free to comment here with additional information and the ticket can be reopened.

closed 2019-02-20 by dirk-thomas

tfoote, 2019-02-20:
Note that default reporting periods are between once every 4 seconds and once every 64 seconds with the goal of 10 to 100 samples per window. https://github.com/ros/ros_comm/blob/acd831359ea245c523d1d12449f52bdac41de229/clients/rospy/src/rospy/impl/statistics.py#L72-L79 https://github.com/ros/ros_comm/blob/acd831359ea245c523d1d12449f52bdac41de229/clients/roscpp/src/libros/statistics.cpp#L46-L50 and they are adjustable by parameter if you'd like to change them.

wentz89, 2019-02-21:
Thanks for that hint.

But anyway the window show there are more than 6300 evaluated elements per statistics-msg (see Delivered in the Plot).
I just took a look on my other data plots and there are connections that only evaluate 1 or 2 messages on one connection per statistics message (the topic frequency was about 1 Hz).
Also the update frequency for the most publisher-subscriber - connections is 1 Hz.

wentz89, 2019-02-21:
So I looked up the source code.
The calculation of the publish_frequency: 

`    
    // dynamic window resizing
    
    if (stats.arrival_time_list.size() > static_cast<size_t>(max_elements) && pub_frequency_ * 2 <= max_window)

    {
      pub_frequency_ *= 2;
    }
    if (stats.arrival_time_list.size() < static_cast<size_t>(min_elements) && pub_frequency_ / 2 >= min_window)

    {
      pub_frequency_ /= 2;
    }
`
So basically if there are more messages then max_elements increase the frequency or if there are less messages then min_elements decrease the frequency. This is fine.

And the decision if one should publish a new statistics message:

    // should publish new statistics?
    if (stats.last_publish + ros::Duration(pub_frequency_) < received_time)
    {
    ...
    }
I recognized that pub_frequency_ is a double value, which doesn't make sense. So a simple example:
stats.last_publish = 1
pub_frequency_ = 10
received_time = 2
Obviously in this case one should publish cause 1 second passed and our frequency is 10 Hz. 
But the if-statement evaluates to: 1 + 10 < 2 -> false, which means don't publish. So this line need to be changed to:

    if (stats.last_publish + ros::Duration(1 / pub_frequency_) < received_time)

@dirk-thomas or do i miss something?

wentz89, 2019-02-21:
I created a pull request: https://github.com/ros/ros_comm/pull/1625

wentz89, 2019-02-22:
@tfoote @dirk-thomas will it be reopened and fixed? I would appreciate cause i am using the topic_statistics quite often.

kikass13, 2019-02-22:
#1621 thanks, we found the same bug at the same time , lel

dirk-thomas, 2019-02-22:
> will it be reopened and fixed?

Since you already opened the pull request #1625 I don't think this ticket needs to be reopened.