ros/ros_comm #1644: Rosbag duration can finish too early

eirik-strand, 2019-03-04:
I have not done a lot of tests with this issue, but I think it can be reproduced. I have a lot of topics, and will not record more than 1s of data. If it takes more than 1s to start the subscribers, everything just hangs. I guess that check_duration finishes before all subscribers got something on the topic. 

If you have a a long list of topics, doing:
`rosbag record --all --duration="0.1s" `
Will cause it to just hang and never finish
if not I guess you can try `0.01s `

cwecht, 2019-03-05:
I was able to reproduce this bug. The hangig is caused by a deadlock involving Recorders queue_mutex_ and TopicManagers advertised_topics_mutex_) caused by calling ros::shutdown in Recorder::checkDuration during the queue_mutex_ lock.

cwecht, 2019-03-05:
Ok, I was wrong about that, it is even worse: the deadlock involves TopicManagers::advertised_topics_mutex TopicManagers::subs_mutex_.

dirk-thomas, 2019-03-08:
Addressed by #1645.

closed 2019-03-08 by dirk-thomas