ros/ros_comm #1654: ros::Time or double?

nslay, 2019-03-13:
This is subjective feedback, but I honestly think this is a design issue and I hope you consider it in ROS2.

First and foremost, I appreciate the hard work of the ROS developers for developing the various tools and APIs for ROS! 

So here's my issue in brief: ros::Time and ros::TimeBase are exception-happy when they shouldn't be.

Not everything done with timestamps deserves the scrutiny of exception handling. I understand the good intention behind, for example, putting exception handling into ros::Time arithmetic, but I don't appreciate programs artificially crashing hours/days into processing because some trivial non-important ros::Time arithmetic expression caused a 32-bit difference exception. I don't care! Much like I don't care if ordinary integer arithmetic overflows in every piece of code I write in much the same way (unless I actually care and explicitly do my own checks)! I don't see why anyone should have to think "Should this harmless difference be wrapped with a try/catch?" I don't need that level of scrutiny everywhere, and if I do, I am going to check the ros::Time's values explicitly anyway!

So what do I do? What you didn't intend! I bypass those checks for those non-critical calculations where wrong calculations don't matter to me at all, and I turn them into double with ros::Time::toSec(). I do it almost everywhere in these situations because who knows what else ros::Time will throw and for what! I really hate that, but I can rest assured that my stuff won't crash artificially because someone gave me a ros::Time(0) or something like that. And I am *not* wrapping every timestamp calculation in a try/catch block... that's dumb! Nobody does this with integers and ros::Time is basically a pair of integers. You're being overprotective of us by checking that things don't overflow and throwing exceptions if they might/do!

Lastly, I observe that some of the sources of those exceptions have some interesting code and comments... like special wrapper functions to throw the exception to prevent code bloat in templates from all the exception stuff compilers generate. The fact that you even need to consider this possibility tells me that maybe it doesn't belong there in boiler plate code like operator- (it doesn't in my opinion).

Oh, and also ros::Time::isValid() is an ambiguously named function. I know what it does, and I also know what I thought it did when I first saw it! Anyway... that's nitpicking. Thanks for reading!

dirk-thomas, 2019-03-13:
> I honestly think this is a design issue and I hope you consider it in ROS2.

The time primitives in ROS 2 are already past the design phase. They have already been implemented quite some time ago.

> ros::Time and ros::TimeBase are exception-happy when they shouldn't be.

Please enumerate the exact cases you are referring to. The blank statement that exceptions don't make sense anywhere doesn't seem helpful.

> I don't appreciate programs artificially crashing hours/days into processing because some trivial non-important ros::Time arithmetic expression caused a 32-bit difference exception. I don't care!

While for your use case you might not care other users don't want a bug like that go unnoticed and prefer to get an exception - either to handle it in the source or to get notified of the problem and fix the problem with the input data.

> I bypass those checks for those non-critical calculations where wrong calculations don't matter to me at all, and I turn them into double with ros::Time::toSec().

As long as you are aware that using a `double` for that will result in a loss of accuracy - again you might not care about the precision in your use case but generally that is not acceptable.

> You're being overprotective of us by checking that things don't overflow and throwing exceptions if they might/do!

Please feel free to propose an alternative which isn't "overprotective" but still allows users to rely on the time functions to be "safe". Removing all kind of exceptions and just ignoring the apparent problem is likely not going to cut it for the majority of the users.

> I observe that some of the sources of those exceptions have some interesting code and comments... like special wrapper functions to throw the exception to prevent code bloat in templates from all the exception stuff compilers generate.

Please feel free to propose specific changes in a pull request. The description alone is a bit too fuzzy to understand what exact part you are talking about.

> ros::Time::isValid() is an ambiguously named function.

Please feel free to propose a new name for a function like this. It would be easy to add a function with a better name beside the existing function. Removing/renaming the existing function isn't likely going to happen since it would imply breaking existing code for no apparent benefit.

dirk-thomas, 2019-03-13:
Since the ticket was mostly a set of questions and personal feedback and there is nothing directtly which should be acted on I will close the ticket for now. Please feel free to continue commenting on it. If there are specific items to be done for this repo I am happy to reopen the ticket (or create a more narrow follow up ticket).

closed 2019-03-13 by dirk-thomas

nslay, 2019-03-13:
Thanks for reading my feedback. This was all feedback and no questions. You can ignore me all you want. I will continue to do all my non-critical timestamp calculations for whatever reason, debugging, logging or debugging information of whatever by turning them into double with ros::Time::toSec(). I don't like it, but I think a poor decision was made to not trust the API users and do all the checks inside ros::Time and throw exceptions in places I believe they don't belong, like subtraction (imagine the nightmares if substracting two integers could throw exceptions). Because you do that, I can't trust ros::Time everywhere and I do it with doubles. I'm sure I'm not the only one.

The alternative is to *not throw exceptions* in arithmetic and silently overflow like the rest of the language. Undefined behavior? I don't care most of the time! Unless I really do care and then I do careful checks... like everyone else working with integers! ROS takes the liberty of doing the check in boiler plate calculations for you and then crashes your program even if you didn't ask for that kind of scrutiny. Don't throw the exceptions in arithmetic and behave like the rest of the language.