ros/ros_comm #1658: Confusing: C++ waitForService timeout is milisec, Python wait_for_service timeout is seconds

peci1, 2019-03-15:
The function docs say it, but it still seems to me confusing that the C++ part takes in milisec and Python seconds.
C++ has another method that takes in `ros::Duration` as timeout. However, Python only accepts doubles. 

I understand that it's impossible to change either behavior of the C++ or the Python calls regarding doubles, but could the Python version try to parse the argument as `rospy.Duration`?

dirk-thomas, 2019-03-15:
Allowing to pass a duration sounds feasible. Please consider to provide a pull request for this feature idea.

peci1, 2019-03-15:
Ok, I'll do the Python PR. 

I also had the idea of deprecating the int/double APIs. What's your view on that?

dirk-thomas, 2019-03-15:
Deprecating existing API in use (since there is no other API) for no apparent reason other than "niceness" isn't something I would merge. Sorry.

peci1, 2019-03-15:
Okay, no problem, I understand that. But when this issue is resolved, there will be the ros::Duration alternative API both in Python and C++. Maybe for N turtle?

dirk-thomas, 2019-03-15:
> Maybe for N turtle?

The benefit needs to clearly outweigh the effort. And honestly I don't see any benefit for existing code to change.

peci1, 2019-03-15:
;)

peci1, 2019-04-26:
Fix published in https://github.com/ros/ros_comm/pull/1703 .

closed 2019-08-07 by dirk-thomas