ros/ros_comm #1664: roslaunch cli does not support remapping arguments

beetleskin, 2019-03-19:
The roslaunch cli should support the same remapping arguments as rosrun. 

The current workaround is to use `args` and then remap in the launch-file (see e.g. https://answers.ros.org/question/55461/does-roslaunch-support-topic-remapping-from-command-line/). This breaks the design idea behind the whole remapping and namespace policy (http://wiki.ros.org/ROS/Patterns/Communication#When_Topics_Should_be_Remapped).

dirk-thomas, 2019-08-07:
Please consider to contribute the feature.

beetleskin, 2019-08-20:
I'll try, but is it even worth the effort given that 2020 is going to be the last ROS1 release? I fear it will never make it to being distributed via a stable release.

closed 2025-05-31 by sloretz