ros/ros_comm #1671: rosgraph/test_names.py: test_load_mappings uses illegal remaps?

cwecht, 2019-03-22:
`test_names.py` contains this line:

    assert {'a': '1', 'b': '2', 'c': '3'} == load_mappings(['c:=3', 'c:=', ':=3', 'a:=1', 'b:=2'])

If I understand the wiki page about [Remapping Argument](http://wiki.ros.org/Remapping%20Arguments) correctly, these are illegal mappings.
These are neither remaps of names, because '1', '2' and '3' are no legal names as specified [here](http://wiki.ros.org/Names), nor they are parameter assignments, because they lack the leading underscore.

Currently this basically does not matter, but in order of my attempts to fix #1459 I'd like to create a function, which performs more extensive checks, to determine if a given string is a remap, than to check, if the string contains ':='. With this test present, this function can only check, if the substring in front of ':=' is a valid name; the later substring must basically be ignored.

cwecht, 2019-04-01:
`test_rospy_client.py` in `test_rospy` does something similar: it requires `rospy.myargv()` to strip out `-bar:=foo` even though it should strip out remaps only.

dirk-thomas, 2019-08-07:
What is the status of this ticket in relation to the merged PR #1683?

cwecht, 2019-08-08:
This is resolved by #1683.

closed 2019-08-08 by cwecht