ros/ros_comm #1672: running ROS on a single CPU core

karim174, 2019-03-22:
is it possible to run all ROS operation on a single core ? the drive for such question is to parallelize the simulation of multiple robots separately on different cores for reinforcement learning distributed algorithms

dirk-thomas, 2019-03-22:
I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2019-03-22 by dirk-thomas

karim174, 2019-03-22:
[https://answers.ros.org/question/319273/assigning-ros-nodes-located-in-a-launch-file-to-a-certain-cpu-core/](url)