ros/ros_comm #1676: rospy.search_param only finds parameters within current namespace

corot, 2019-03-26:
Reading the [wiki documentation](http://wiki.ros.org/rospy/Overview/Parameter%20Server), I understand that _rospy.search_param('param')_ will search the entire parameters set until finding 'param', but that's not the case: it will only explore current's node namespace.

Key source line is [this](https://github.com/ros/ros_comm/blob/melodic-devel/clients/rospy/src/rospy/msproxy.py#L162):
`self.target.searchParam(rospy.names.get_caller_id(), key)`
We pass current node name to searchParam. If I pass instead the node name containing 'param', it finds it.

Is this a bug? Or am I misinterpreting the documentation?

cwecht, 2019-04-01:
@corot could you provide a minimal (not) working example?

corot, 2019-04-07:
Example? Well, just call rospy.search_param with any parameter present in the param server and it won't find it:

`#!/usr/bin/env python
import rospy

if __name__ == '__main__':
    rospy.init_node("test_search_param")
    print rospy.search_param('update_frequency')
    rospy.spin()
`
_update_frequency_ is one of costmap's parameters, and search_param won't find it.

cwecht, 2019-04-12:
Well, if i do this:

1. start  ``roscore``
2. set the parameter: ``rosparam set /update_frequency 10``
3. run your test programm: ``./test_param.py``

I get ``/update_frequency`` printed out. It would be nice, if you could provide a more precise/concrete MWE.

corot, 2019-04-12:
Yup, sorry, looks like root ns works.  Try instead:

rosparam set /foo/update_frequency 10

You will get None

cwecht, 2019-04-16:
The documentation of ``search_param`` is correct but somewhat misleading. ``search_param`` takes the given name, appends it to the nodes namespace and checks, if this parameter exists. If not, it proceeds by appending the given name to next higher namespace and checks, if this parameter exists, and so on.

``/foo/update_frequency`` is not found, because ``update_frequency`` does not reside in a namespace, in which the node resides. The documentation is not absolutely clear about this.

Nevertheless this approach has the benefit, that the result is unambiguous. If ``search_param`` should be allowed to find ``/foo/`update_frequency``, what if ``/bar/update_frequency`` also exists? With the current interface this is a problem.

corot, 2019-04-23:
> If not, it proceeds by appending the given name to next higher namespace 
this sentence is what makes me expect that it will search for the parameter in the whole parameter server, as the highest namespace is /. True you can get several results, but that also can happen if the calling node has both `<node>/foo/update_frequency` and `<node>/bar/update_frequency` parameters (no idea what will happen in that case... I suppose `search_param` will return the first one he finds)
So the functions should return a list in that cases!.
Would you accept a PR on melodic with the behavior I propose? Otherwise I would change the doc to make it clear that `search_param` only searches within calling node namespace

cwecht, 2019-04-23:
> if the calling node has both `<node>/foo/update_frequency` and `<node>/bar/update_frequency` parameters (no idea what will happen in that case... I suppose `search_param` will return the first one he finds)

If I do this:

1.  start ``roscore``
2.  set the parameter: ``rosparam set /test_search_param/foo/update_frequency 10``
3.  run your test programm: ``./test_param.py``

the paramter will not be found.

Actually ``test_param.py`` can only find ``/update_frequency`` and ``/test_search_param/update_frequency``.

Maybe the documentation should give a concrete example:

The node
~~~
#!/usr/bin/env python
import rospy

if __name__ == '__main__':
    rospy.init_node("bar")
    print rospy.search_param('param')
    rospy.spin()
~~~

started via ``./test.py __ns:=/foo``

can find these parameters (and in this order):

~~~
/foo/bar/param
/foo/param
/param
~~~

corot, 2019-04-24:
Right. If that's the desired behavior, we can close this issue. But still, documentation is confusing, imho

cwecht, 2019-04-24:
Ok, I've just updated the [wiki-page](http://wiki.ros.org/rospy/Overview/Parameter%20Server#Searching_for_parameter_keys).

closed 2019-04-24 by corot