ros/ros_comm #1677: roscpp service call doesn't return error message when service fails

yli-cpr, 2019-03-26:
In rospy, service providers can raise `rospy.ServiceException` with error messages. However roscpp clients cannot get the error message. When you work with some roscpp nodes and rospy nodes, this can be a problem. If the error message is delivered through TCP (it should), so roscpp should be able to extract the message and return to caller.

cwecht, 2019-04-30:
@yli-cpr could you elaborate on the concrete problem you are facing? Do you need an error message for debugging/logging purposes or do you want your nodes to be able to recover form different errors in different ways?

If you change the logger-level of ``ros.roscpp.roscpp_internal`` to ``Debug`` yo usould get more information.

yli-cpr, 2019-05-01:
@cwecht I just want to log the error message. There are ways to work around it, like including the error message in response. But I am just saying there is a gap between `rospy` and `roscpp`. Here is the code where `rospy` client reads the `rospy.ServiceException` error message: https://github.com/ros/ros_comm/blob/melodic-devel/clients/rospy/src/rospy/impl/tcpros_service.py#L365

And it is called when the `OK` byte is 0: https://github.com/ros/ros_comm/blob/melodic-devel/clients/rospy/src/rospy/impl/tcpros_service.py#L332

So it is supported by the protocol, but `roscpp` doesn't provide the service error.

cwecht, 2019-05-06:
Well, it kind of does, the message is read here: https://github.com/ros/ros_comm/blob/29053c4832229efa7160fb944c05e3bc82e11540/clients/roscpp/src/libros/service_server_link.cpp#L246 and printed here: https://github.com/ros/ros_comm/blob/29053c4832229efa7160fb944c05e3bc82e11540/clients/roscpp/src/libros/service_server_link.cpp#L378

yli-cpr, 2019-05-06:
Good to know. But in some cases, client wants to handle the error string by itself, for example: mapping to error messages in another language for globalization, or forward the error message to web client, so on. So it would be nice the `call()` method returns the string to caller.

mkatliar, 2022-02-05:
> So it would be nice the `call()` method returns the string to caller.

Or throw an exception to the caller, if the service also threw.

closed 2025-05-31 by sloretz