ros/ros_comm #1689: [feature request] [rosbag] Read arguments from the parameter server

rtimpe, 2019-04-05:
We've run into the issue where we'd like to configure a rosbag record node created in a roslaunch file using a yaml config file, rather than command line arguments.  Our current node looks something like this

```
<node name="rosbag" pkg="rosbag" type="record"
        args="
          /topic1
          /topc2
          /topic3"/>
```

It would be nice if we could get the same effect with something like
```
<node name="rosbag" pkg="rosbag" type="record">
  <rosparam file="$(arg rosbag_config_yaml)"/>
</node>
```

where the config file would look like
```
topics:
  - '/topic1'
  - '/topic2'
  - '/topic3'
```
This would make it easy to record different topics in different configurations of the launch file.  From what I can tell the recommended way of doing this currently is by using ```<arg>``` tags in the launch file, but this suffers from the limitation that they cannot be populated from a separate yaml file.  Logically, it seems as if rosbag nodes should be able to read parameters from the parameter server.

mikepurvis, 2019-04-05:
This can be considered for upstream rosbag, but note that many organizations which are running persistent rosbag recording end up wrapping rosbag's C++ API in their own manager-type node. This is exceedingly easy to do and may end up giving you more flexibility to implement the kinds of specific recording policies which you require:

https://github.com/ros/ros_comm/blob/ac2f29d75bd0b8fd983939ea634b03c4549d2363/tools/rosbag/include/rosbag/recorder.h#L88-L118

rtimpe, 2019-04-05:
Thanks for the response. Yes, it's likely that we'll end up wrapping the C++ API.  We were just wondering if it was possible to continue using the CLI in the meantime since it mostly fits our needs well.  I'll take a closer look at the C++ API though.  It sounds like writing a wrapper may be easier than I'd thought.

dirk-thomas, 2019-08-07:
Closing since the C++ API is better suited for that kind of use cases.

closed 2019-08-07 by dirk-thomas