ros/ros_comm #1697: Python Import Error from rospack

maspe36, 2019-04-12:
Hi peeps,

I'm following along with the ROS tutorials ([Creating a Package](http://wiki.ros.org/ROS/Tutorials/CreatingPackage)) and when I run `rospack depends1 beginner_tutorials`

I get this error
```
dev@dev-test-eth1:~/catkin_ws/src$ rospack depends1 beginner_tutorials
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/__init__.py", line 40, in <module>
    from .installers import InstallerContext, Installer, \
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 36, in <module>
    from rospkg.os_detect import OsDetect
  File "/usr/lib/python2.7/dist-packages/rospkg/__init__.py", line 41, in <module>
    from .manifest import InvalidManifest, Manifest, parse_manifest_file
  File "/usr/lib/python2.7/dist-packages/rospkg/manifest.py", line 40, in <module>
    import xml.dom.minidom as dom
  File "/home/dev/miniconda2/lib/python2.7/xml/dom/minidom.py", line 22, in <module>
    from xml.dom.xmlbuilder import DOMImplementationLS, DocumentLS
  File "/home/dev/miniconda2/lib/python2.7/xml/dom/xmlbuilder.py", line 3, in <module>
    import copy
  File "/home/dev/miniconda2/lib/python2.7/copy.py", line 52, in <module>
    import weakref
  File "/home/dev/miniconda2/lib/python2.7/weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/__init__.py", line 40, in <module>
    from .installers import InstallerContext, Installer, \
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 36, in <module>
    from rospkg.os_detect import OsDetect
  File "/usr/lib/python2.7/dist-packages/rospkg/__init__.py", line 41, in <module>
    from .manifest import InvalidManifest, Manifest, parse_manifest_file
  File "/usr/lib/python2.7/dist-packages/rospkg/manifest.py", line 40, in <module>
    import xml.dom.minidom as dom
  File "/home/dev/miniconda2/lib/python2.7/xml/dom/minidom.py", line 22, in <module>
    from xml.dom.xmlbuilder import DOMImplementationLS, DocumentLS
  File "/home/dev/miniconda2/lib/python2.7/xml/dom/xmlbuilder.py", line 3, in <module>
    import copy
  File "/home/dev/miniconda2/lib/python2.7/copy.py", line 52, in <module>
    import weakref
  File "/home/dev/miniconda2/lib/python2.7/weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref
[rospack] Error: could not find python module 'rosdep2.rospack'. is rosdep up-to-date (at least 0.10.4)?
```

I have a miniconda environment configured and I verified that rospack is indeed using the correct interpreter (Edited the /usr/lib/python2.7/dist-packages/rosdep2/__init__.py file to print sys.executable). I can import these modules 

```
dev@dev-test-eth1:~/catkin_ws/src$ python
Python 2.7.15 |Anaconda, Inc.| (default, Dec 14 2018, 19:04:19) 
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from rosdep2 import rospack
>>> import weakref
>>> from _weakref import _remove_dead_weakref
>>> 
```

Python Interpreter
```
dev@dev-test-eth1:~/catkin_ws/src$ which python
/home/dev/miniconda2/bin/python
```

Pip 
```
dev@dev-test-eth1:~/catkin_ws/src$ which pip
/home/dev/miniconda2/bin/pip
```

Located rosdep2
```
dev@dev-test-eth1:~/catkin_ws/src$ locate rosdep2
/usr/lib/python2.7/dist-packages/rosdep2
/usr/lib/python2.7/dist-packages/rosdep2/__init__.py
/usr/lib/python2.7/dist-packages/rosdep2/__init__.pyc
/usr/lib/python2.7/dist-packages/rosdep2/_version.py
/usr/lib/python2.7/dist-packages/rosdep2/_version.pyc
/usr/lib/python2.7/dist-packages/rosdep2/catkin_packages.py
/usr/lib/python2.7/dist-packages/rosdep2/catkin_packages.pyc
/usr/lib/python2.7/dist-packages/rosdep2/catkin_support.py
/usr/lib/python2.7/dist-packages/rosdep2/catkin_support.pyc
/usr/lib/python2.7/dist-packages/rosdep2/core.py
/usr/lib/python2.7/dist-packages/rosdep2/core.pyc
/usr/lib/python2.7/dist-packages/rosdep2/dependency_graph.py
/usr/lib/python2.7/dist-packages/rosdep2/dependency_graph.pyc
/usr/lib/python2.7/dist-packages/rosdep2/gbpdistro_support.py
/usr/lib/python2.7/dist-packages/rosdep2/gbpdistro_support.pyc
/usr/lib/python2.7/dist-packages/rosdep2/installers.py
/usr/lib/python2.7/dist-packages/rosdep2/installers.pyc
/usr/lib/python2.7/dist-packages/rosdep2/loader.py
/usr/lib/python2.7/dist-packages/rosdep2/loader.pyc
/usr/lib/python2.7/dist-packages/rosdep2/lookup.py
/usr/lib/python2.7/dist-packages/rosdep2/lookup.pyc
/usr/lib/python2.7/dist-packages/rosdep2/main.py
/usr/lib/python2.7/dist-packages/rosdep2/main.pyc
/usr/lib/python2.7/dist-packages/rosdep2/model.py
/usr/lib/python2.7/dist-packages/rosdep2/model.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms
/usr/lib/python2.7/dist-packages/rosdep2/rep3.py
/usr/lib/python2.7/dist-packages/rosdep2/rep3.pyc
/usr/lib/python2.7/dist-packages/rosdep2/rosdistrohelper.py
/usr/lib/python2.7/dist-packages/rosdep2/rosdistrohelper.pyc
/usr/lib/python2.7/dist-packages/rosdep2/rospack.py
/usr/lib/python2.7/dist-packages/rosdep2/rospack.pyc
/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py
/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.pyc
/usr/lib/python2.7/dist-packages/rosdep2/shell_utils.py
/usr/lib/python2.7/dist-packages/rosdep2/shell_utils.pyc
/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py
/usr/lib/python2.7/dist-packages/rosdep2/sources_list.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/__init__.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/__init__.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/arch.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/arch.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/cygwin.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/cygwin.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/debian.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/debian.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/freebsd.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/freebsd.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/gem.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/gem.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/gentoo.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/gentoo.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/opensuse.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/opensuse.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/osx.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/osx.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/pip.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/pip.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/redhat.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/redhat.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/slackware.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/slackware.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/source.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/source.pyc
```

Python Path
```
dev@dev-test-eth1:~/catkin_ws/src$ echo $PYTHONPATH 
/opt/ros/kinetic/lib/python2.7/dist-packages:/usr/lib/python2.7/dist-packages
```

Some additional info 
- Ubuntu 16.04
- Kinetic ROS dist
- miniconda2 py2.7

I'm super confused, any ideas?

dirk-thomas, 2019-04-12:
I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

Also you are using miniconda which is not part of the recommended setup.

closed 2019-04-12 by dirk-thomas

mikepurvis, 2019-04-12:
For miniconda to work, you'd need to build the entire ROS workspace from scratch, see: http://wiki.ros.org/Installation/Source

As soon as you use the OSRF-supplied packages, you're locked into using whatever `/usr/bin/python` is on your system, as there a bunch of packages which link to it in various ways (especially rospack and roslib, and anything which links to them).

maspe36, 2019-04-12:
Thanks for the info and direction guys, I appreciate the patience!

For any googlers, here is the link to the ROS question https://answers.ros.org/question/320995/python-import-error-from-rospack/