ros/ros_comm #1713: roscpp client segfaults on shutdown

igor-semenov, 2019-05-07:
Got segmentation fault on exit, stacktrace is:
```
* frame #0: 0x00007fde488e636e libstdc++.so.6`std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) + 46

frame #1: 0x00007fde49701465 libroscpp.so`ros::TransportPublisherLink::initialize(boost::shared_ptr<ros::Connection> const&) + 1253

frame #2: 0x00007fde49702192 libroscpp.so`ros::TransportPublisherLink::onRetryTimer(ros::SteadyTimerEvent const&) + 1106

frame #3: 0x00007fde497069eb libroscpp.so`ros::TimerManager<ros::SteadyTime, ros::WallDuration, ros::SteadyTimerEvent>::TimerQueueCallback::call() + 379

frame #4: 0x00007fde496d1a20 libroscpp.so`ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) + 1184

frame #5: 0x00007fde496d2e93 libroscpp.so`ros::CallbackQueue::callAvailable(ros::WallDuration) + 1459

frame #6: 0x00007fde497155d4 libroscpp.so`ros::internalCallbackQueueThreadFunc() + 100```

igor-semenov, 2019-05-07:
Looks like here: https://github.com/ros/ros_comm/blob/29053c4832229efa7160fb944c05e3bc82e11540/clients/roscpp/src/libros/transport_publisher_link.cpp#L91
we don't check whether weak_ptr<> lock is successful.

igor-semenov, 2019-05-07:
https://github.com/ros/ros_comm/pull/1714

dirk-thomas, 2019-08-07:
Addressed by #1714.

closed 2019-08-07 by dirk-thomas