ros/ros_comm #1716: message_filters for gazebo_msgs::ContactsState

crankler, 2019-05-12:
Does anyone use bumper on ros?
I use message_filters for imu + joint_states + force_sensor, but it can't enter callback,
can anybody tell me how to do? Thans you for your help!~
Here is my question:
https://answers.ros.org/question/322951/synchronize-imu-joint_states-foot_sensor-errors/

cwecht, 2019-05-13:
Could you provide an example rosbag? That would be really helpful!

crankler, 2019-05-13:
Yes, Thank you for your help! here is my bag:

[subset.bag.tar.gz](https://github.com/ros/ros_comm/files/3173119/subset.bag.tar.gz)

cwecht, 2019-05-13:
The time stamps of the ``ContactsState`` messages are always zero in this rosbag.

crankler, 2019-05-13:
But when I run  `rostopic echo /leg1_4_bumper`, it can show me  the time,
and can you tell me how can I get the stamps?
I just use it form urdf setting.
Thank you!

cwecht, 2019-05-13:
Well, if I run ``rostopic echo -b subset.bag /leg1_4_bumper | less``, I get something like this:
~~~
header: 
  seq: 371
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: "leg1_4_link"
...
~~~

under Ubuntu 18.04 with melodic. The field ``secs`` ``nsecs`` of ``stamp`` are relevant.

I don't know how to fix this issue with gazebo, but this seems to be an issue with gazebo or with your setup.

crankler, 2019-05-14:
But when I only just use message filter for 4 foot sensors, it can enter the callback, and it can give me a 
frequency of 30hz, you can also try it.
Maybe it's the bumper error, I will try other foot sensor simulation plug and then to test it.

cwecht, 2019-05-14:
Well, ``TimeSynchronizer`` fire a callback as soon as there are messages of each subscriber, which have the same time-stamp. Since every ``ContactsState``s time stamp is zero, it will fire regulary. The ``ApproximateTime`` policy does a similar thing.

crankler, 2019-05-14:
Yes, I try these two ways as you do.
Today I write a timestamp from bumper topic, I echo and find there have a info string  which show time, 
so I make it as the real time stamp.
At now, it can use message_filter, but it will show a warning as flowing:
[ WARN] [1557848077.068834211]: Messages of type 3 arrived out of order (will print only once)

cwecht, 2019-05-15:
Well, if you provide a rosbag, which reproduces this issue, I can have a look at this.

crankler, 2019-05-16:
This error occur by chance, so I can't find where can cause this error.
But in all, Thank you for your help!!!

cwecht, 2019-05-17:
You're welcome! But I think, it is pretty clear now, that this is not an issue of message_filters, so this issue can be closed.

closed 2019-05-17 by crankler