ros/ros_comm #1718: Publisher unable to connect to a topic when multiprocessing is used.

thebhatman, 2019-05-17:
I have a function which is being run by two multiprocessing processes. This function publishes to a topic using rospy publisher. But the publisher is unable to connect to the topic. pub.get_num_connections() is zero and I also added a sleep time to wait till the publisher topic connection happens. But it never connected. I needed help with how I could publish using multiprocessing (using the same publisher). Has anyone faced this problem before?

cwecht, 2019-05-20:
@thebhatman could you provide a minimal (not) working example?

joshs333, 2019-06-06:
We got the same error on a project I am working on, we only saw it after the run was over and we were analyzing bag data and have been unable to produce.

iljastas, 2020-09-30:
Is there a solution?

peci1, 2021-09-15:
I think the explanation is simple - multiprocessing spawns actual new processes. And the publisher you created is a process-only object which cannot be shared with other processes. Create a new publisher in each subprocess (and probably also call `rospy.init_node()` in each)

closed 2025-05-31 by sloretz