ros/ros_comm #1719: Communicate with ros through XML-RPC

amranov, 2019-05-23:
Hi guys, I am working on a project where i need to communicate to ROS( i plan motion for robot arm using moveit) from NVIDIA (i get command from the robot hand got to pose using XMLRPC python server) to issues some command to the hand. 
My question is: how could  i run on the server and pass XMLRPC requests on to the hand? 
I am not sure if its the right place to ask this request, but i will appreciate any help and guidance.

fujitatomoya, 2019-05-23:
i believe that you can ask this kind of stuff via https://answers.ros.org/questions/

not sure what you are trying to do, you want to communicate with rosmaster with XML-RPC?
if so you already have via ros_comm to handle the configuration data for topic and so on.
why do you not just use topic instead for your requirement?

amranov, 2019-05-23:
@fujitatomoya  thank you for quick reply,
actually i want to communicate my robot arm with robot hand , the robot hand envirement NVIDIA with no ROS installed, i created XMLRPC server to issue command 
`import sys
sys.path.append('/home/nvidia/rmstudio/lib') 

from xmlrpc.server import SimpleXMLRPCServer
import xmlrpc.client
from xmlrpc.server import SimpleXMLRPCRequestHandler
import time
import base64
import netifaces as ni
import numpy as np
import sys
import traceback

#import rmlib
#rm = rmlib.RMLib()

class RequestHandler(SimpleXMLRPCRequestHandler):
    rpc_paths = ('/RPC2')

    def do_OPTIONS(self):
        self.send_response(200)
        self.end_headers()

    # Add these headers to all responses
    def end_headers(self):
        self.send_header("Access-Control-Allow-Headers",
        "Origin, X-Requested-With, Content-Type, Accept")
        self.send_header("Access-Control-Allow-Origin", "*")
        SimpleXMLRPCRequestHandler.end_headers(self)

def run_cmd(dat_string):
    print(dat_string)
    try:
        print(exec(dat_string))
    except SyntaxError as err:
        error_class = err.__class__.__name__
        detail = err.args[0]
        line_number = err.lineno
        print("%s at line %d: %s" % (error_class, line_number, detail))
        time.sleep(0.15) 
        return("%s at line %d: %s" % (error_class, line_number, detail))
    except Exception as err:
        error_class = err.__class__.__name__
        detail = err.args[0]
        cl, exc, tb = sys.exc_info()
        line_number = traceback.extract_tb(tb)[-1][1]
        print("%s at line %d: %s" % (error_class, line_number, detail))
        time.sleep(0.15) # add this so that the GUI can catch the last block at which the error happened
        return("%s at line %d: %s" % (error_class, line_number, detail))
    else: 
        return "ok"

def init():
    try:
        rm
    except:
        import rmlib
        global rm
        rm = rmlib.RMLib()
#    import rmlib
#    global rm
#    rm = rmlib.RMLib()
    return True

def irimage():
	import cv2
	import numpy as np
	rm.set_laser_state(False)
	rm.set_auto_exposure_on(True)
	#rm.set_exposure(165000)
	
	img = rm.get_ir_image()
	img = cv2.flip(img,0)
	img = cv2.flip(img,1)
	img = np.array(img*255./img.max()).astype(np.uint8)

	cv2.imwrite('/media/ramdisk/snapshot.png',img)
	return True	

def listToPose(l):
  assert type(l) is list
  return {'x' : l[0], 'y' : l[1], 'z' : l[2], 'rx' : l[3], 'ry' : l[4], 'rz' : l[5]}

def get_object_defs():
        #...
        print("Object definitions requested")
        return "M3 Screw%Banana%Ratstail%Apple%Newby%No_underscore%Complex screw, 110#3%Not so nice long string%And one last one"

def get_object_pose(object):
	# your stuff
        pose = [0.563, -0.114, 0.22, 2.20, 2.223,0]
        return listToPose(pose)

print("Starting XML-RPC server on port 8101")
server = SimpleXMLRPCServer(("", 8101),requestHandler=RequestHandler)
server.register_introspection_functions()
server.register_function(run_cmd)
server.register_function(init)
server.register_function(irimage)
server.register_function(get_object_defs)
server.register_function(get_object_pose)
server.serve_forever()`

planners topics are published in ROS (moveit)
but my question is how to call this server from ROS (AS i told i am new in ROS ) and send the planners command to the hand ?

fujitatomoya, 2019-05-24:
are you asking you want to send some request from robot arm(ROS) to hand(Non ROS) via XML-RPC?
if so, that XML-RPC communication is not related to ROS, am i right? Basically ROS(ros_comm) does not have any XML-RPC base APIs for ROS application, in fact that ROS uses XML-RPC internally to control the configuration data.

amranov, 2019-05-24:
@fujitatomoya  yes i want to send request from arm (ROS) to hand (No ROS) via XML-RPC.
the communication is not related to ROS, but is it possible to setup XMLRPC python client from ROS to send request to the XML-RPC server on the hand ?

fujitatomoya, 2019-05-24:
No, i do not think so as far as i know. besides in this situation, XMLRPC python server IS NOT actually ROS system.
so after all, this is NOT related about ROS, even ros_comm. i think that it is the question about common xml-rpc client, so this is not the appropriate place we talk about that.

tfoote, 2019-05-28:
I would recommend looking for the recommended platform for the XMLRPC project to ask questions or try something more generic like stack overflow. 

We ask that ROS questions be asked on https://answers.ros.org But as this isn't a ROS question it's not appropriate there either.

closed 2019-05-28 by tfoote