ros/ros_comm #1728: roslaunch fails to show stdout/stderr from remote nodes

mferenduros, 2019-06-14:
Hi,
We've had frequent problems with stdout/stderr not being forwarded from nodes launched remotely using &lt;machine&gt;.

We isolated it to a simple case where two nodes are launched on a remote machine. One node crashes soon after launch printing a stack-trace and the second continues running, but the stack-trace is not seen on the local machine.
If the second node is removed from the launchfile, the stack-trace appears as expected on the local machine.

I notice that [remoteprocess.py / is_alive](https://github.com/ros/ros_comm/blob/89ca9a7bf288b09d033a1ab6966ef8cfe39e1002/tools/roslaunch/src/roslaunch/remoteprocess.py#L250) says that it needs to be called regularly to drain the SSH buffer, but AFAICT it is only called at launch-time. Could this be the problem?

dirk-thomas, 2019-08-07:
Please check if #1267 already addressed your problem.

dirk-thomas, 2019-10-03:
Closing due to no response.

closed 2019-10-03 by dirk-thomas

peci1, 2021-09-09:
I think the problem is not in not calling is_alive - I see it regularly called. The problem might be the size of the read buffer, which is hardcoded to 2048 bytes. So until 2048 characters are written to stderr, nothing is read. I assume that if you have only the one crashing node, the remote roslaunch will stop and close connection, which can probably also trigger a read of the output buffer.

peci1, 2021-09-09:
Have a look at our monkey-patch of roslaunch - https://github.com/ctu-vras/scoped_roslaunch . We set the buffers there to 128 bytes and make some more polishing of the output of remote nodes (including printing stdout, too!)