ros/ros_comm #1729: Missing data

getupgetup, 2019-06-17:
I'm writing a 3d-slam project. The project needs to subscrible PointCloud2 topic and then processes it. I use a subscriber to subscribe the data published by "rosbag play xx.bag". I count the number of times when I entered my callback function.Then I find the number is smaller than the amount of messages in the bag I play. It means some data is missed. Can anyone help me solve that? Thanks~

cwecht, 2019-06-18:
This problem can have many reasons. Did you have a look the CPU load? Maybe your node processes the callback to slowly. What queue size in the subscriber do you have?

getupgetup, 2019-06-19:
> This problem can have many reasons. Did you have a look the CPU load? Maybe your node processes the callback to slowly. What queue size in the subscriber do you have?
My CPU is Intel® Core™ i7-6700 CPU @ 3.40GHz × 8.One of eight CPUs is 100% in some preriod when running the programme.I set the queue size to 0 to avoid dropping data.Is there anything wrong?Thank you!

christophebedard, 2019-06-20:
> I set the queue size to 0 to avoid dropping data.Is there anything wrong?

If the queue size is 0, it *will* drop data if it can't keep up with the message rate. Try setting it to 10, and then if data is still dropped, increase it.

getupgetup, 2019-06-20:
> > I set the queue size to 0 to avoid dropping data.Is there anything wrong?
> 
> If the queue size is 0, won't it drop data if it can't keep up with the message rate? Maybe try setting it to 10, and then if data is still dropped, increase it.

According to [http://wiki.ros.org/roscpp/Overview/Publishers%20and%20Subscribers](url), the queue size is infinite if I set it to 0. Could you please tell me why the subscriber drops data?Thank you!

christophebedard, 2019-06-20:
Oh. I admit I did not know that! Looking at `SubscriptionQueue::push()` that does seem to be the case.

Just to check, can you try using a very big queue size?

getupgetup, 2019-06-20:
> Oh. I admit I did not know that! Looking at `SubscriptionQueue::push()` that does seem to be the case.
> 
> Just to check, can you try using a very big queue size?

I tried setting the queue size larger than the amount of messages in my bag. There was still some data discarded.

cwecht, 2019-06-20:
What happens, if you set the rate of the playback very low? Like `rosbag play -r 0.01`?

getupgetup, 2019-06-21:
> What happens, if you set the rate of the playback very low? Like `rosbag play -r 0.01`?

Nope. I've test it by using different rates from 0.5 to 5.0. There were few times that data was not discarded.

getupgetup, 2019-06-21:
> What happens, if you set the rate of the playback very low? Like `rosbag play -r 0.01`?

I test with "rosbag play -r 0.1" and it seems that data won't be discarded! Actually, the PointCloud2 topic is published about every 100ms by default and my callback function cost is about 50ms on average. What's the problem? Thank you~

cwecht, 2019-06-28:
ok, what happens, if you try to record another rosbag from the published data (with the normal rate)?  Does this work? It might also be helpful, if you could provide the rosbag.

getupgetup, 2019-07-26:
> ok, what happens, if you try to record another rosbag from the published data (with the normal rate)? Does this work? It might also be helpful, if you could provide the rosbag.

You can get the bag from https://github.com/RobustFieldAutonomyLab/jackal_dataset_20170608.It's called "same_position.bag" in folder "same_start_end_position".
Ps:sorry for my late response

cwecht, 2019-08-03:
Ok, the rosbag seems to be valid. Could you create a [Minimal, Reproducible Example](https://stackoverflow.com/help/minimal-reproducible-example) of you problem? If not possible, the actual node should also be fine.

dirk-thomas, 2019-09-16:
Closing until there is a reproducible example.

closed 2019-09-16 by dirk-thomas