ros/ros_comm #1734: Pyhton's roslaunch API TypeError: not all arguments converted during string formatting

AchmadFathoni, 2019-07-02:
I am trying python roslaunch with argument by following the [wiki](http://wiki.ros.org/roslaunch/API%20Usage) using this code:

```
#!/usr/bin/python
import roslaunch

uuid = roslaunch.rlutil.get_or_generate_uuid(None, False)
roslaunch.configure_logging(uuid)

cli_arg = ['hw_interface','test.launch','save_path:=hello']
roslaunch_file = roslaunch.rlutil.resolve_launch_arguments(cli_arg)

roslaunch_args = cli_arg[2:]
launch_file = [(roslaunch_file, roslaunch_args)]

parent = roslaunch.parent.ROSLaunchParent(uuid, launch_file)
parent.start()
```
but it return error:

> ... logging to /home/tony/.ros/log/0e38ee00-9c7a-11e9-990f-68a3c454c001/roslaunch-T410-26775.log
> Traceback (most recent call last):
>   File "/home/tony/lidar_ws/src/hw_interface/script/test.py", line 16, in <module>
>     parent.start()
>   File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
>     self._start_infrastructure()
>   File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
>     self._load_config()
>   File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
>     roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
>   File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 450, in load_config_default
>     logger.info('loading config file %s'%f)
> TypeError: not all arguments converted during string formatting

`test.launch` is empty and can be launched by roslaunch command in terminal. I am using ROS Kinetic. Some person already post it [here](http://answers.ros.org/question/318117/python-api-for-roslaunch-with-parameter/) but there still no answer.

dirk-thomas, 2019-08-07:
Please provide a reproducible example. `hw_interface` isn't a ROS package available as a Debian package in Melodic.

Achllle, 2019-10-09:
@dirk-thomas This is true for any launch file. The example given in section 3.1 of the [wiki](http://wiki.ros.org/roslaunch/API%20Usage) with `turtle_tf2`:
```
#!/usr/bin/python

import rospy
import roslaunch

uuid = roslaunch.rlutil.get_or_generate_uuid(None, False)
roslaunch.configure_logging(uuid)

cli_args = ['turtle_tf2', 'turtle_tf2_demo.launch']
roslaunch_args = cli_args[2:]  # empty but doesn't affect the outcome
roslaunch_file = [(roslaunch.rlutil.resolve_launch_arguments(cli_args)[0], roslaunch_args)]

parent = roslaunch.parent.ROSLaunchParent(uuid, roslaunch_file)
parent.start()
rospy.sleep(10)  # prevent from immediately shutting down
```
This also fails for a launch file that has arguments (couldn't immediately find one in the Debian repository). You can get this to work for launch files that don't have arguments by just supplying the resolved launch file path:
```
roslaunch_file = roslaunch.rlutil.resolve_launch_arguments(cli_args)
```
instead of the list of the tuple. I couldn't get it to work with arguments yet.

Achllle, 2019-10-10:
Digging a bit deeper into roslaunch, it seems like there might be a bug here.
The documentation of [load_config_default()](https://github.com/ros/ros_comm/blob/6b1ee1a43b5b0a970d85612435e269818c89a152/tools/roslaunch/src/roslaunch/config.py#L410) in `roslaunch/config.py` states:
```
    @param roslaunch_files: list of launch files to load. Each item may also
      be a tuple where the first item is the launch file and the second item
      is a string containing arguments.
    @type  roslaunch_files: [str|(str, str)]
```
This definition is a little bit ambiguous as the `@param` tag specifies the second element in each tuple as a string while the `@type` tag seems to indicate it's a tuple of strings (which makes more sense).
When using the tuple definition `[(path1, (arg1, arg2)), (path2, (arg1, arg2)]`, which means changing the line in the OP's example and wiki from
```
roslaunch_args = cli_arg[2:]
```
to
```
roslaunch_args = tuple(cli_arg[2:])
```
so that it is a tuple, I still get the same error as the OP.

I don't know if this is a bug or if the docstring is off.

Achllle, 2019-10-23:
@dirk-thomas would you mind at least removing the more information needed tag? Thank you.

anug7, 2019-11-16:
Facing the same issues. Any updates on this?

Achllle, 2019-12-09:
From the wiki: 

> The code API of roslaunch is very unstable and should not be used directly. In order to support the new features that are being planned, it may be necessary to make major, incompatible changes to the programmatic API.

The SIG for roslaunch hasn't seen updates since 2014 so I'm assuming devs are putting effort in roslaunch2 instead.

Achllle, 2019-12-09:
This was fixed in lunar by #1115 . I confirmed the fix works for Kinetic.

Achllle, 2019-12-09:
As stated in #1431 there are no plans to backport this into Kinetic. 

@dirk-thomas Since this is a bug and the fix seems simple and considering Kinetic is still supported, could you confirm whether it makes sense to open a PR to backport? What is the recommended approach for Kinetic users?

RDaneelOlivav, 2020-01-02:
Having this issue in kinetic still:

self.uuid = roslaunch.rlutil.get_or_generate_uuid(None, False)
            cli_args = [self._rospackage_name, self._launch_file_name, 'x:=0.0', 'z:=2.0']
            roslaunch_args = tuple(cli_args[2:])
            roslaunch_file = [(roslaunch.rlutil.resolve_launch_arguments(cli_args)[0], roslaunch_args)]
            self.launch = roslaunch.parent.ROSLaunchParent(self.uuid, roslaunch_file)
            self.launch.start()

Any insights?

Achllle, 2020-01-02:
@RDaneelOlivav it doesn't seem like you read the comment above

> As stated in #1431 there are no plans to backport this into Kinetic.

RDaneelOlivav, 2020-01-07:
Yeh did read it but I thought that there might have been some changes sorry for the comment then. I'll work on melodic or future versions, thanks.

closed 2025-05-31 by sloretz