ros/ros_comm #1736: sockaddr_in.in transport_udp.cpp is not initialized

danielwangksu, 2019-07-02:
Coverity report Using uninitialized value "sin". Field "sin.sin_zero" is uninitialized when calling "connect".
https://github.com/ros/ros_comm/blob/29053c4832229efa7160fb944c05e3bc82e11540/clients/roscpp/src/libros/transport/transport_udp.cpp#L160

I remember the uninitialized sin_zero might cause undefined behavior, not sure if it is still valid

cwecht, 2019-07-04:
[Reading from an uninitialized value is UB](https://www.learncpp.com/cpp-tutorial/uninitialized-variables-and-undefined-behavior/). `sin_zero` is somewhat strange: People say you can [ignore](http://www.ccplusplus.com/2011/10/struct-sockaddrin.html) it or that it [does not matter](https://www.gta.ufrj.br/ensino/eel878/sockets/sockaddr_inman.html), if you do not initialize it. it seems, that it has been removed from [POSIX](https://pubs.opengroup.org/onlinepubs/009695399/basedefs/netinet/in.h.html) (scroll to changelog). For [windows](https://docs.microsoft.com/de-de/windows/win32/api/ws2def/ns-ws2def-sockaddr_in) it seems, that it should be initialized. [This german book](http://openbook.rheinwerk-verlag.de/linux_unix_programmierung/Kap11-005.htm#RxxKap110050400038A1F01B1A4) (sorry for that) says, that using `sin_zero` should be avoided, as it is not part of POSIX, but recommends to initialize the whole struct, which is reasonable.

danielwangksu, 2019-07-10:
I think it is the same for "server_address_". It seems never get initialized:
https://github.com/ros/ros_comm/blob/29053c4832229efa7160fb944c05e3bc82e11540/clients/roscpp/src/libros/transport/transport_udp.cpp#L249

PR #1740 (2019-08-07): roscpp/transport_udp: zero-initialize sockaddr_in object
Fixed #1736.
closed 2019-08-07 by dirk-thomas