ros/ros_comm #1742: Rosbags consume memory (leak?)

harlowja, 2019-07-06:
So looking at the bag.h it appears that when a bag is opened in write mode only that every single entry that is written gets an index entry written into ``connection_indexes_`` which means that every single message (or nearly every one, depending on timing) will cause ``connection_indexes_`` to get bigger and bigger and bigger (even though it will never be used in write only mode); This eventually (for large bags) grows to a very large ``connection_indexes_`` size (which may or may not take up a large portion of memory for something that is never used in write only mode in the first place).

Affected lines seem to be:

https://github.com/ros/ros_comm/blob/lunar-devel/tools/rosbag_storage/include/rosbag/bag.h#L585-L586

It's easy to see this problem happen, open a bag (or multiple) and then write to them in a loop (from some other rosbag playing messages); eventually if u have a long enough bag to play you will just continue getting new messages into ``connection_indexes_`` and if the bag is big enough you can outrun available memory...

dirk-thomas, 2019-08-07:
I guess the fact that the index if kept in memory for efficient access is by design. Please consider to contribute if you think this should be improved.

harlowja, 2019-08-07:
Except in write-only mode, there isn't any access in the first place :-/

Arnatious, 2019-08-22:
Can confirm that the only usage of `connection_indexes_` is in `Bag::readConnectionIndexRecord200()`.

Should be a simple matter to only append to it when `mode_` is `Read`. I'll get on it.

dirk-thomas, 2019-10-03:
Assuming #1798 addressed the problem I will close this ticket. Please feel free to continue commenting and if necessary the ticket can be reopened.

closed 2019-10-03 by dirk-thomas