ros/ros_comm #1758: rostopic: list --host raises ValueError

kakkoko, 2019-07-14:
Steps to reproduce:
1. Execute `roscore` on a terminal
2. Execute `rostopic list --host` on another terminal

Expected:
```
Host [vboxres]:
  /rosout
  /rosout_agg
```

Actual:
```
Host [vboxros]:
Traceback (most recent call last):
  File "/opt/ros/melodic/bin/rostopic", line 35, in <module>
    rostopic.rostopicmain()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostopic/__init__.py", line 2119, in rostopicmain
    _rostopic_cmd_list(argv)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostopic/__init__.py", line 2059, in _rostopic_cmd_list
    exitval = _rostopic_list(topic, verbose=options.verbose, subscribers_only=options.subscribers, publishers_only=options.publishers, group_by_host=options.hostname) or 0
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostopic/__init__.py", line 1242, in _rostopic_list
    verbose, indent='  ')
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostopic/__init__.py", line 1150, in _sub_rostopic_list
    topics = list(set([t for t, _, _ in pubs] + [t for t, _, _ in subs]))                
ValueError: need more than 2 values to unpack
```

Note `rostopic list` worked as expected:
```
$ rostopic list
/rosout
/rosout_agg
```

Environment:
- Ubuntu 18.04.2 LTS
- ros-melodic-desktop-full/bionic,now 1.4.1-0bionic.20190618.145748 amd64
    * installed according to http://wiki.ros.org/melodic/Installation/Ubuntu

artivis, 2019-08-01:
I can confirm the issue.
Ubuntu 18.04
rostopic 1.14.3

danielwangksu, 2019-08-07:
the same, confirm the issue
Ubuntu 18.04
rostopic 1.14.3

PR #1780 (2019-08-07): rostopic: _rostopic_list_group_by_host: publisher/subscriber lists ha…
…ve correct type now

This fixes #1758.
closed 2019-08-07 by dirk-thomas