ros/ros_comm #1762: Dead loop if accept connection error in XmlRpcServer

randoms, 2019-07-17:
If there is an error when accepting connections, `_accept_error` will be set to [true](https://github.com/ros/ros_comm/blob/melodic-devel/utilities/xmlrpcpp/src/XmlRpcServer.cpp#L196), and `XmlRpcServer` will be [add to XmlRpcDispatch](https://github.com/ros/ros_comm/blob/melodic-devel/utilities/xmlrpcpp/src/XmlRpcServer.cpp#L166). but when accept connection is OK  `_accept_error` is not set to true, and `XmlRpcServer` will still be add to `XmlRpcDispatch` in each work loop.

dirk-thomas, 2019-08-07:
Please provide reproducible steps for the problem you are describing.

randoms, 2019-08-08:
First edit `acceptConnection` in `XmlRpcServer.cpp` to simulate a network error.

```cpp
XmlRpcServer::acceptConnection()
{
  // simulate network error
  static int error_count = 0;
  error_count += 1;
  int s = XmlRpcSocket::accept(this->getfd());
  printf("XmlRpcServer::acceptConnection: socket %d\n", s);
  XmlRpcUtil::log(2, "XmlRpcServer::acceptConnection: socket %d", s);
  // network error at the first time
  // if (s < 0)
  if(error_count < 2)
  {
    printf("XmlRpcServer::acceptConnection: Could not accept connection (%s).\n", XmlRpcSocket::getErrorMsg().c_str());
    //this->close();
    XmlRpcUtil::error("XmlRpcServer::acceptConnection: Could not accept connection (%s).", XmlRpcSocket::getErrorMsg().c_str());

    // Note that there was an accept error; retry in 1 second
    _accept_error = true;
    _accept_retry_time_sec = _disp.getTime() + ACCEPT_RETRY_INTERVAL_SEC;
    return 0; // Stop monitoring this FD
  }
```

add some output message to `XmlRpcServer::work`

```cpp
void 
XmlRpcServer::work(double msTime)
{
  XmlRpcUtil::log(2, "XmlRpcServer::work: waiting for a connection");
  if(_accept_error && _disp.getTime() > _accept_retry_time_sec) {
    printf("_disp.getTime() %f _accept_retry_time_sec %f \n", _disp.getTime(), _accept_retry_time_sec);
    printf("add source\n");
    _disp.addSource(this, XmlRpcDispatch::ReadableEvent);
  }
  _disp.work(msTime);
}
```

Compile a simple message publisher with this lib. 
Simple publisher
```cpp
#include "ros/ros.h"
#include "std_msgs/String.h"
#include <ros/console.h>

int main(int argc, char **argv)
{
	ros::init(argc, argv, "talker");
	ros::NodeHandle n;
	ros::Publisher chatter_pub = n.advertise<std_msgs::String>("chatter", 1000);
	ros::Rate loop_rate(1);
	int count = 0;
	while (ros::ok())
	{
		std_msgs::String msg;
		std::stringstream ss;
		ss << "hello world " << count;
		msg.data = ss.str();
		std::cout << "send message: hello world" << std::endl;
		chatter_pub.publish(msg);
		ros::spinOnce();
		loop_rate.sleep();
		++count;
	}
	return 0;
}
```

And here is the output of the simple message publisher

```
send message: hello world
XmlRpcServer::acceptConnection: socket 10
XmlRpcServer::acceptConnection: Could not accept connection (Success).
send message: hello world
_disp.getTime() 13945.746811 _accept_retry_time_sec 13945.746811 
add source
XmlRpcServer::acceptConnection: socket 11
_disp.getTime() 13945.847382 _accept_retry_time_sec 13945.746811 
add source
_disp.getTime() 13945.947540 _accept_retry_time_sec 13945.746811 
add source
_disp.getTime() 13946.047719 _accept_retry_time_sec 13945.746811 
add source
_disp.getTime() 13946.147883 _accept_retry_time_sec 13945.746811 
add source
_disp.getTime() 13946.248028 _accept_retry_time_sec 13945.746811 
...
```

`XmlRpcServer` will be added to `XmlRpcDispatch` in each work loop, until system run out of resources.

The fix is simple , set `_accept_error` to false when connected successfully.

```cpp
  else  // Notify the dispatcher to listen for input on this source when we are in work()
  {
    _accept_error = false; // set to false, when connected
    XmlRpcUtil::log(2, "XmlRpcServer::acceptConnection: creating a connection");
    _disp.addSource(this->createConnection(s), XmlRpcDispatch::ReadableEvent);
  }
  return XmlRpcDispatch::ReadableEvent; // Continue to monitor this fd
```

danielwangksu, 2019-08-08:
confirm this issue

randoms, 2019-08-13:
Add pulll request #1791

PR #1791 (2019-10-03): Fix dead loop if accept connection error in XmlRpcServer
Fix #1762
closed 2019-10-03 by dirk-thomas