ros/ros_comm #1774: catkin_make error

yassinebara, 2019-07-23:
Hi,
When I run catkin_make, I get the following error :
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "octomap_msgs" with
  any of the following names:

    octomap_msgsConfig.cmake
    octomap_msgs-config.cmake

  Add the installation prefix of "octomap_msgs" to CMAKE_PREFIX_PATH or set
  "octomap_msgs_DIR" to a directory containing one of the above files.  If
  "octomap_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Anyone can help please.
Thanks

cwecht, 2019-07-23:
How did you get octomag_msgs? From the deb-repositories?

artivis, 2019-08-01:
I'm afraid this issue is not `ros_comm` related. You may want to expose your problem again on [answers.ros.org](https://answers.ros.org/questions/) to get some help.
@mikepurvis I think this can be closed.

dirk-thomas, 2019-08-07:
I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2019-08-07 by dirk-thomas

yhdurkaya, 2019-12-25:
`sudo apt install ros-melodic-octomap-ros`
This will fix your problem.

BolinZho, 2020-04-07:
"sudo apt install ros-melodic-octomap-ros" does work.

Ajilforoushan, 2022-08-31:
for me works with:

$ sudo apt-get install ros-melodic-octomap

Ajilforoushan, 2022-08-31:
> Hi, When I run catkin_make, I get the following error : CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package): Could not find a package configuration file provided [...]

===============
for me works with:

$ sudo apt-get install ros-melodic-octomap