ros/ros_comm #1800: Python rosbag::Bag read_messages function does not accept return_connection_header

ricber, 2019-09-03:
Hi,
I am launching this script:
```
import rosbag

with rosbag.Bag('2019-05-24-09-53-28-high-density-imurenamed.bag', 'w') as outbag:
    for topic, msg, t, conn_header in rosbag.Bag('2019-05-24-09-53-28-high-density.bag').read_messages(return_connection_header=True):
        if topic == "/f746zg_imu/data_raw" or topic == "/f746zg_imu/mag":
            msg.header.frame_id = "nucleo_imu"
        outbag.write(topic, msg, t, connection_header=conn_header)
```

and I obtain the following error:

```
Traceback (most recent call last):
  File "rename_imu.py", line 4, in <module>
    for topic, msg, t, conn_header in rosbag.Bag('2019-05-24-09-53-28-high-density.bag').read_messages(return_connection_header=True):
TypeError: read_messages() got an unexpected keyword argument 'return_connection_header'

```

ricber, 2019-09-03:
The "return_connection_header" parameter is not available in the kinetic branch.

dirk-thomas, 2019-10-03:
The feature was added in #1372 for Melodic and in general features are not backported to older ROS distro. Therefore I am closing this ticket.

closed 2019-10-03 by dirk-thomas