ros/ros_comm #1803: Error when using lambda instead of boost::bind in TimeSynchronizer

aPonza, 2019-09-16:
kinetic, ubuntu 16.04, gcc 5.5.0

MWE comes directly from the [TimeSynchronizer example](http://wiki.ros.org/message_filters#Time_Synchronizer)
```
#include <message_filters/subscriber.h>
#include <message_filters/time_synchronizer.h>
#include <sensor_msgs/Image.h>
#include <sensor_msgs/CameraInfo.h>

using namespace sensor_msgs;
using namespace message_filters;

void callback(const ImageConstPtr& image, const CameraInfoConstPtr& cam_info)
{
  // Solve all of perception here...
}

int main(int argc, char** argv)
{
  ros::init(argc, argv, "vision_node");

  ros::NodeHandle nh;

  message_filters::Subscriber<Image> image_sub(nh, "image", 1);
  message_filters::Subscriber<CameraInfo> info_sub(nh, "camera_info", 1);
  TimeSynchronizer<Image, CameraInfo> sync(image_sub, info_sub, 10);
  sync.registerCallback([](const ImageConstPtr& image_msg,
            const CameraInfoConstPtr& caminfo_msg)
        {
          callback(image_msg, caminfo_msg);
        });

  ros::spin();

  return 0;
}
```

results in

```
In file included from /usr/include/boost/bind.hpp:22:0,
                 from /opt/ros/kinetic/include/ros/publisher.h:35,
                 from /opt/ros/kinetic/include/ros/node_handle.h:32,
                 from /opt/ros/kinetic/include/ros/ros.h:45,
                 from /opt/ros/kinetic/include/message_filters/subscriber.h:38,
                 from /home/ap/catkin_ws/src/cobotica/pickable_localizer/src/pickable_localizer.cpp:347:
/usr/include/boost/bind/bind.hpp: In instantiation of ‘struct boost::_bi::result_traits<boost::_bi::unspecified, main(int, char**)::<lambda(const ImageConstPtr&, const CameraInfoConstPtr&)> >’:
/usr/include/boost/bind/bind.hpp:883:48:   required from ‘class boost::_bi::bind_t<boost::_bi::unspecified, main(int, char**)::<lambda(const ImageConstPtr&, const CameraInfoConstPtr&)>, boost::_bi::list9<boost::arg<1>, boost::arg<2>, boost::arg<3>, boost::arg<4>, boost::arg<5>, boost::arg<6>, boost::arg<7>, boost::arg<8>, boost::arg<9> > >’
/opt/ros/kinetic/include/message_filters/signal9.h:281:52:   required from ‘message_filters::Connection message_filters::Signal9<M0, M1, M2, M3, M4, M5, M6, M7, M8>::addCallback(C&) [with C = const main(int, char**)::<lambda(const ImageConstPtr&, const CameraInfoConstPtr&)>; M0 = sensor_msgs::Image_<std::allocator<void> >; M1 = sensor_msgs::CameraInfo_<std::allocator<void> >; M2 = message_filters::NullType; M3 = message_filters::NullType; M4 = message_filters::NullType; M5 = message_filters::NullType; M6 = message_filters::NullType; M7 = message_filters::NullType; M8 = message_filters::NullType]’
/opt/ros/kinetic/include/message_filters/synchronizer.h:310:40:   required from ‘message_filters::Connection message_filters::Synchronizer<Policy>::registerCallback(const C&) [with C = main(int, char**)::<lambda(const ImageConstPtr&, const CameraInfoConstPtr&)>; Policy = message_filters::sync_policies::ExactTime<sensor_msgs::Image_<std::allocator<void> >, sensor_msgs::CameraInfo_<std::allocator<void> >, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_filters::NullType>]’
/home/ap/catkin_ws/src/cobotica/pickable_localizer/src/pickable_localizer.cpp:373:10:   required from here
/usr/include/boost/bind/bind.hpp:69:37: error: no type named ‘result_type’ in ‘struct main(int, char**)::<lambda(const ImageConstPtr&, const CameraInfoConstPtr&)>’
     typedef typename F::result_type type;
                                     ^
```

I'm trying to figure out a synchronized callback for a pair of class members, and wanted to use a lambda to follow modern style (C++11 is supported afaik in kinetic) but this error pops up. I couldn't manage to wrap the lambda in a way that works to make a `result_type` trait available (I have some links but I don't think that's the point). I'll probably resort to `boost::bind` and leave a TODO for myself, but wanted to check if this is intended behaviour or I'm messing something up. I don't have any other instance that wasn't compatible with lambdas in my codebase, and no other instance of `bind` and `function` (both `boost::` or `std::`) used.

aPonza, 2019-09-17:
Reading through the code, this comes from using `boost::bind` in message filters (which requires `result_type` like the compiler says), and that must likely be done to keep compatibility with C++03, so this sounds like a "wontfix" close.

I tried to wrap the lambda yesterday to add the trait and failed (mainly following [this](https://stackoverflow.com/a/7943765/9926122)/[this](https://stackoverflow.com/a/2565037/9926122)), so unless anybody has suggestions this can be closed.

rcywongaa, 2020-10-09:
You can get around the `result_type` complaint by providing the template arguments for `registerCallback`
```
     sync.registerCallback<std::function<void(const ImageConstPtr&, const CameraInfoConstPtr&)>>(
     [](const ImageConstPtr& image_msg, const CameraInfoConstPtr& caminfo_msg)
    { ... }
```
However, this brings up another error
```
/usr/include/boost/bind/bind.hpp:833:9: error: no matching function for call to object of type 'std::function<void (const boost::shared_ptr<const sensor_msgs::Image_<std::allocator<void> > > &, const boost::shared_ptr<const sensor_msgs::CameraInfo_<std::allocator<void> > > &)>'
        unwrapper<F>::unwrap(f, 0)(a[base_type::a1_], a[base_type::a2_], a[base_type::a3_], a[base_type::a4_], a[base_type::a5_], a[base_type::a6_], a[base_type::a7_], a[base_type::a8_], a[base_type::a9_]);
        ^~~~~~~~~~~~~~~~~~~~~~~~~~
```
I couldn't get `registerCallback` to use the correct template

BruceChanJianLe, 2021-05-28:
Hi @rcywongaa @aPonza not sure if you are still interested in the solution.

I believe you will need to specify all 9 of them in `std::function` like so

```cpp
     sync.registerCallback<std::function<void(
                const ImageConstPtr&,
                const CameraInfoConstPtr&,
                const boost::shared_ptr<const message_filters::NullType>,
                const boost::shared_ptr<const message_filters::NullType>,
                const boost::shared_ptr<const message_filters::NullType>,
                const boost::shared_ptr<const message_filters::NullType>,
                const boost::shared_ptr<const message_filters::NullType>,
                const boost::shared_ptr<const message_filters::NullType>,
                const boost::shared_ptr<const message_filters::NullType>
     )>>(
     [](const ImageConstPtr& image_msg, const CameraInfoConstPtr& caminfo_msg, ...)
    { ... }
```

This compile for me no issue. Haven't tested yet.

closed 2025-05-31 by sloretz