ros/ros_comm #1825: Issue with operator= and Publisher/Subscriber instances under GCC 9

mikepurvis, 2019-10-04:
We're beginning to plan our Ubuntu 20.04 migration, and the following innocent-looking code is blowing up under GCC 9:

```cpp
  ros::Publisher  di_pub;
  ros::Subscriber do_sub;

  [...]

  di_pub = gpio_nh.advertise<gpio_msgs::GpioPins>("/digital_io", 10, true);
  do_sub = gpio_nh.subscribe("/cmd_digital_io", 10, &Node::digital_output_cb, &n);
```

The error is:

```
error: implicitly-declared ros::Publisher& ros::Publisher::operator=(const ros::Publisher&) is deprecated
error: implicitly-declared ros::Subscriber& ros::Subscriber::operator=(const ros::Subscriber&) is deprecated
```

@iwanders pointed out the `Publisher` and similar classes in roscpp have explicit constructors, destructors, and copy constructors but are missing move construction, move assignment, and copy assignment:

https://github.com/ros/ros_comm/blob/melodic-devel/clients/roscpp/include/ros/publisher.h#L51

This is in violation of the rule of 6: https://github.com/isocpp/CppCoreGuidelines/blob/master/CppCoreGuidelines.md#c21-if-you-define-or-delete-any-default-operation-define-or-delete-them-all

Would a PR be welcome to rectify this? Given that the master branch is now targeting Melodic and C++14, there should be no need to protect any of this stuff with ifdef guards. I'm not sure what all classes would be affected, but basically [everything in the roscpp API](https://docs.ros.org/api/roscpp/html/annotated.html) should be checked, especially common handle classes like Subscriber, Timer, SteadyTimer, NodeHandle, ServiceServer, ServiceClient.

mikepurvis, 2020-01-23:
@dirk-thomas We're seeing this coming up in a bunch of repos now; any thoughts about how to best proceed?

dirk-thomas, 2020-02-03:
> Would a PR be welcome to rectify this?

Sure, please go ahead and create a PR.

mikepurvis, 2020-02-19:
Seems to be resolved by #1888.

closed 2020-02-19 by mikepurvis