ros/ros_comm #1830: [py3] test_rostest test node run with wrong python version

sloretz, 2019-10-15:
It looks like `rostest` tests in this repo are using the wrong python version when `ROS_PYTHON_VERSION` is `3`. To reproduce, edit [`test_rostest` `just_advertise`](https://github.com/ros/ros_comm/blob/melodic-devel/test/test_rostest/test/just_advertise) to check the python version.

```diff
sloretz@nxt ~/w/n/s/ros_comm> git diff
diff --git a/test/test_rostest/test/just_advertise b/test/test_rostest/test/just_advertise
index 48507030f..6bda8abe5 100755
--- a/test/test_rostest/test/just_advertise
+++ b/test/test_rostest/test/just_advertise
@@ -44,6 +44,11 @@ import roslib.message
 
 
 if __name__ == '__main__':
+    import os
+    import sys
+    sys.stderr.write(repr(sys.version) + " " + repr(os.environ['ROS_PYTHON_VERSION']) + "\n")
+    assert sys.version[0] == os.environ['ROS_PYTHON_VERSION']
+
     rospy.init_node('just_advertise')
     msg_name = rospy.get_param('~msg_name')
     msg_class = roslib.message.get_message_class(msg_name)
```

In a `noetic` Python 3 workspace (I'm using Buster, but I'd expect the same issue on Bionic), run the tests using Python 3.

```
$ echo $ROS_PYTHON_VERSION
3
$ python3 src/catkin/bin/catkin_make_isolated --pkg test_rostest --cmake-args -DCATKIN_ENABLE_TESTING=1 --make-args run_tests
```

The test node is using python 2

```
'2.7.16 (default, Apr  6 2019, 01:42:57) \n[GCC 8.3.0]' '3'
Traceback (most recent call last):
  File "/home/sloretz/ws/noetic/src/ros_comm/test/test_rostest/test/just_advertise", line 50, in <module>
    assert sys.version[0] == os.environ['ROS_PYTHON_VERSION']
```

<details><summary>Full test output</summary>

```
==> Processing catkin package: 'test_rostest'
==> Building with env: '/home/sloretz/ws/noetic/devel_isolated/test_rosservice/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/home/sloretz/ws/noetic/build_isolated/test_rostest'
==> make run_tests -j8 -l8 in '/home/sloretz/ws/noetic/build_isolated/test_rostest'
Built target tests
Removing test result files from '/home/sloretz/ws/noetic/build_isolated/test_rostest/test_results/test_rostest'
- removing '/home/sloretz/ws/noetic/build_isolated/test_rostest/test_results/test_rostest/rostest-test_publishtest.xml'
- removing '/home/sloretz/ws/noetic/build_isolated/test_rostest/test_results/test_rostest/rosunit-publishtest.xml'
Built target clean_test_results_test_rostest
-- run_tests.py: execute commands
  /usr/bin/python3 /home/sloretz/ws/noetic/src/ros_comm/tools/rostest/scripts/rostest --pkgdir=/home/sloretz/ws/noetic/src/ros_comm/test/test_rostest --package=test_rostest --results-filename test_publishtest.xml --results-base-dir "/home/sloretz/ws/noetic/build_isolated/test_rostest/test_results" /home/sloretz/ws/noetic/src/ros_comm/test/test_rostest/test/publishtest.test 
... logging to /home/sloretz/.ros/log/rostest-43822c0b032c-32069.log
[ROSUNIT] Outputting test results to /home/sloretz/ws/noetic/build_isolated/test_rostest/test_results/test_rostest/rostest-test_publishtest.xml
'2.7.16 (default, Apr  6 2019, 01:42:57) \n[GCC 8.3.0]' '3'
Traceback (most recent call last):
  File "/home/sloretz/ws/noetic/src/ros_comm/test/test_rostest/test/just_advertise", line 50, in <module>
    assert sys.version[0] == os.environ['ROS_PYTHON_VERSION']
AssertionError
[Testcase: testpublishtest] ... ok

[ROSTEST]-----------------------------------------------------------------------

[test_rostest.rosunit-publishtest/test_publish][passed]

SUMMARY
 * RESULT: SUCCESS
 * TESTS: 1
 * ERRORS: 0
 * FAILURES: 0

rostest log file is in /home/sloretz/.ros/log/rostest-43822c0b032c-32069.log
-- run_tests.py: verify result "/home/sloretz/ws/noetic/build_isolated/test_rostest/test_results/test_rostest/rostest-test_publishtest.xml"
Built target _run_tests_test_rostest_rostest_test_publishtest.test
Built target _run_tests_test_rostest_rostest
Built target _run_tests_test_rostest
Built target run_tests
```
</details>

In this specific case it doesn't matter if the `just_advertise` node is using Python 2 since that's not what is being tested. However, this represents a larger problem with `rostest` and Python. `rostest` [requires the test node to be executable](https://github.com/ros/ros/blob/2b307e1854ebf3954a5b4110f891278f555e79a6/core/roslib/src/roslib/packages.py#L387-L404), and the [shebang in `src` space says `python`](https://github.com/ros/ros_comm/blob/6b1ee1a43b5b0a970d85612435e269818c89a152/test/test_rostest/test/just_advertise#L1). On Buster, `python` is Python 2.

One possible fix is to use `launch-prefix="python$(env ROS_PYTHON_VERSION)"` ([used by `tf2_geometry_msgs`](https://github.com/ros/geometry2/blob/dbc1254d45e71fb072dfb42abc2b9cf1e272ee76/tf2_geometry_msgs/test/test_python.launch#L2)).

```diff
diff --git a/test/test_rostest/test/publishtest.test b/test/test_rostest/test/publishtest.test
index b65e81b51..fbf129b8d 100644
--- a/test/test_rostest/test/publishtest.test
+++ b/test/test_rostest/test/publishtest.test
@@ -11,7 +11,7 @@
   </node>
 
   <node name="talker_2"
-        pkg="test_rostest" type="just_advertise">
+        pkg="test_rostest" type="just_advertise" launch-prefix="python$(env ROS_PYTHON_VERSION)">
     <rosparam>
       msg_name: std_msgs/String
     </rosparam>
```

However, since that was added to `geometry2`, it's been hypothesized this won't work on Windows because `python2` or `python3` commands won't exist. Currently on Windows there is a hard-coded check for files ending in `.py`, but that doesn't explain how the extension-less `just_advertise` runs. If `launch-prefix` does interfere with how Windows picks the python executable, then it will need to be understood to come up with a new workaround.

https://github.com/ros/ros_comm/blob/89ca9a7bf288b09d033a1ab6966ef8cfe39e1002/tools/roslaunch/src/roslaunch/node_args.py#L263-L266

sloretz, 2019-12-03:
I spent a bit of time looking at making `catkin_install_python()` generate a relay script in the devel space similar to what happens to scripts listed in `setup.py`. I did this in a docker container that did not have a `python` or `python2` command.

* https://github.com/sloretz/catkin/tree/devel_python_relay
* https://github.com/sloretz/ros_comm/tree/install_py_test_nodes

With these patches the test appears to pass, but there are a couple issues.

1. The test reports as passing, but something still prints `/usr/bin/env: ‘python’: No such file or directory`. Some python script with an unmodified shebang is being run during the test, and it didn't appear to affect the test results. What is the script being run?
1. Running `rosrun test_rostest talker.py` shows a prompt complaining of non-unique executable. The devel space script does work. Can rosrun be made to just pick the devel space script when there are duplicates?

<details><summary>test_rostest test results with above branches</summary>

```
==> Processing catkin package: 'test_rostest'
==> Building with env: '/home/sloretz/ws/noetic/devel_isolated/test_rosservice/env.sh'
==> cmake /home/sloretz/ws/noetic/src/ros_comm/test/test_rostest -DCATKIN_DEVEL_PREFIX=/home/sloretz/ws/noetic/devel_isolated/test_rostest -DCMAKE_INSTALL_PREFIX=/home/sloretz/ws/noetic/install_isolated -DCATKIN_ENABLE_TESTING=1 -G Unix Makefiles in '/home/sloretz/ws/noetic/build_isolated/test_rostest'
/home/sloretz/ws/noetic/devel_isolated/test_rosservice/env.sh: 1: /home/sloretz/ws/noetic/devel_isolated/roslisp/etc/catkin/profile.d/99.roslisp.sh: python: not found
CMake Warning (dev) at CMakeLists.txt:3 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    CMAKE_PROJECT_VERSION
    CMAKE_PROJECT_VERSION_MAJOR
    CMAKE_PROJECT_VERSION_MINOR
    CMAKE_PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Using CATKIN_DEVEL_PREFIX: /home/sloretz/ws/noetic/devel_isolated/test_rostest
-- Using CMAKE_PREFIX_PATH: /home/sloretz/ws/noetic/devel_isolated/test_rosservice;/home/sloretz/ws/noetic/devel_isolated/test_rospy;/home/sloretz/ws/noetic/devel_isolated/test_rosbag;/home/sloretz/ws/noetic/devel_isolated/stereo_msgs;/home/sloretz/ws/noetic/devel_isolated/map_msgs;/home/sloretz/ws/noetic/devel_isolated/sensor_msgs;/home/sloretz/ws/noetic/devel_isolated/roswtf;/home/sloretz/ws/noetic/devel_isolated/nodelet_topic_tools;/home/sloretz/ws/noetic/devel_isolated/actionlib;/home/sloretz/ws/noetic/devel_isolated/rosbag;/home/sloretz/ws/noetic/devel_isolated/topic_tools;/home/sloretz/ws/noetic/devel_isolated/tf2_py;/home/sloretz/ws/noetic/devel_isolated/test_rosparam;/home/sloretz/ws/noetic/devel_isolated/test_rosmaster;/home/sloretz/ws/noetic/devel_isolated/test_roslaunch;/home/sloretz/ws/noetic/devel_isolated/test_rosgraph;/home/sloretz/ws/noetic/devel_isolated/test_roscpp;/home/sloretz/ws/noetic/devel_isolated/test_rosbag_storage;/home/sloretz/ws/noetic/devel_isolated/test_bond;/home/sloretz/ws/noetic/devel_isolated/rqt_gui_py;/home/sloretz/ws/noetic/devel_isolated/rqt_gui_cpp;/home/sloretz/ws/noetic/devel_isolated/rostopic;/home/sloretz/ws/noetic/devel_isolated/rospy_tutorials;/home/sloretz/ws/noetic/devel_isolated/rosnode;/home/sloretz/ws/noetic/devel_isolated/rosbag_storage;/home/sloretz/ws/noetic/devel_isolated/message_filters;/home/sloretz/ws/noetic/devel_isolated/kdl_parser_py;/home/sloretz/ws/noetic/devel_isolated/joint_state_publisher;/home/sloretz/ws/noetic/devel_isolated/filters;/home/sloretz/ws/noetic/devel_isolated/dynamic_reconfigure;/home/sloretz/ws/noetic/devel_isolated/rosout;/home/sloretz/ws/noetic/devel_isolated/roscpp_tutorials;/home/sloretz/ws/noetic/devel_isolated/nodelet;/home/sloretz/ws/noetic/devel_isolated/bondpy;/home/sloretz/ws/noetic/devel_isolated/bondcpp;/home/sloretz/ws/noetic/devel_isolated/roscpp;/home/sloretz/ws/noetic/devel_isolated/xmlrpcpp;/home/sloretz/ws/noetic/devel_isolated/webkit_dependency;/home/sloretz/ws/noetic/devel_isolated/visualization_msgs;/home/sloretz/ws/noetic/devel_isolated/urdfdom_py;/home/sloretz/ws/noetic/devel_isolated/urdf_parser_plugin;/home/sloretz/ws/noetic/devel_isolated/trajectory_msgs;/home/sloretz/ws/noetic/devel_isolated/tf2_eigen;/home/sloretz/ws/noetic/devel_isolated/tf2_bullet;/home/sloretz/ws/noetic/devel_isolated/tf2;/home/sloretz/ws/noetic/devel_isolated/tf2_msgs;/home/sloretz/ws/noetic/devel_isolated/test_roslib_comm;/home/sloretz/ws/noetic/devel_isolated/std_srvs;/home/sloretz/ws/noetic/devel_isolated/shape_msgs;/home/sloretz/ws/noetic/devel_isolated/rqt_py_common;/home/sloretz/ws/noetic/devel_isolated/rosmsg;/home/sloretz/ws/noetic/devel_isolated/rosgraph_msgs;/home/sloretz/ws/noetic/devel_isolated/nav_msgs;/home/sloretz/ws/noetic/devel_isolated/move_base_msgs;/home/sloretz/ws/noetic/devel_isolated/kdl_conversions;/home/sloretz/ws/noetic/devel_isolated/eigen_conversions;/home/sloretz/ws/noetic/devel_isolated/geometry_msgs;/home/sloretz/ws/noetic/devel_isolated/diagnostic_msgs;/home/sloretz/ws/noetic/devel_isolated/bond;/home/sloretz/ws/noetic/devel_isolated/actionlib_msgs;/home/sloretz/ws/noetic/devel_isolated/std_msgs;/home/sloretz/ws/noetic/devel_isolated/smclib;/home/sloretz/ws/noetic/devel_isolated/rqt_web;/home/sloretz/ws/noetic/devel_isolated/rqt [... cut ...]
Scanning dependencies of target run_tests
Built target run_tests
<== Finished processing package [146 of 160]: 'test_rostest'
```
</details>

<details><summary>rosrun output with devel space relay</summary>

```
$ rosrun test_rostest talker.py
[rosrun] You have chosen a non-unique executable, please pick one of the following:
1) /home/sloretz/ws/noetic/devel_isolated/test_rostest/lib/test_rostest/talker.py
2) /home/sloretz/ws/noetic/src/ros_comm/test/test_rostest/test_nodes/talker.py
#? 
```
</details>

sloretz, 2020-01-02:
`catkin_install_python()` installing a relay script into the devel space doesn't completely solve the problem for tests. First, calling `catkin_install_python()` on test nodes means they get installed, which is not desired. Second, some test nodes try to import python scripts in the same directory as them.

Next idea, maybe this calls for a new CMake macro? `catkin_devel_python()`? It would create the devel-space relay with a hook to prepend the location of the script to `sys.path` so it can import files next to it, but would not install to the install space.

mikepurvis, 2020-01-02:
It's goofy, but I wonder about prepending the `PATH` on those test invocations so that `/usr/bin/env python` routes to a one-off wrapper script that execs `$PYTHON_EXECUTABLE` or `python$ROS_PYTHON_VERSION`.

Note that this only applies to that exact shebang— anything going to `/usr/bin/python` will still be broken, but it has the nice effect that test nodes can also be hardcoded to `/usr/bin/env python3` and that will bypass the wrapper. The wrapper will only kick in for packages which are purporting to support 2 and 3 from a single source (basically mirroring what happens with the shebang-rewriting magic found in `catkin_install_python`).

It does leave a bit of a dangling question about what the long term here should be. Ideally we'd have an approach that limits awkwardness during the transitional period and has none at all once everything is python3-only.

sloretz, 2020-01-02:
> It's goofy, but I wonder about prepending the PATH on those test invocations so that /usr/bin/env python routes to a one-off wrapper script that execs `$PYTHON_EXECUTABLE` or `python$ROS_PYTHON_VERSION`.

If I understand correctly, it looks like this could be done by modifying [create_node_process()](https://github.com/ros/ros_comm/blob/d93fcae779f5eb9a62dba5d7e9210ef635f8dd38/tools/roslaunch/src/roslaunch/nodeprocess.py#L105). It would need to [find the node again](https://github.com/ros/ros_comm/blob/d93fcae779f5eb9a62dba5d7e9210ef635f8dd38/tools/roslaunch/src/roslaunch/node_args.py#L253) so it could read the shebang at the top of the file. `roslaunch` would need to have a directory containing an executable named `python`. Then `roslaunch` could prepend `PATH` with that directory in the environment given to the node it's launching.

Somewhat similar to this is the code `roslaunch` uses to call `python` on windows. It assumes a lack of extension (no `.exe`) means it uses python, which is an assumption that doesn't hold outside of windows so I don't think it can be reused.

https://github.com/ros/ros_comm/blob/d93fcae779f5eb9a62dba5d7e9210ef635f8dd38/tools/roslaunch/src/roslaunch/node_args.py#L266-L270

On the topic of `shebang`'s, yet another similar idea is to add a ROS package [python2or3](https://github.com/sloretz/python2or3/pull/1/files) that executes whatever was given by `ROS_PYTHON_VERSION`. It avoids modifying catkin/roslaunch, but requires every python script shebang to be changed to `/usr/bin/env python2or3`.

> It does leave a bit of a dangling question about what the long term here should be. Ideally we'd have an approach that limits awkwardness during the transitional period and has none at all once everything is python3-only.

:man_shrugging: There isn't much of a transition period since Noetic is a hard switch to Python 3, so it's not clear to me that long term will be any different from the short term.

For install space and debian packages it seems easy: follow PEP-394 recommendations by using `#!/usr/bin/env python` and make sure it gets rewritten when installed. Standalone scripts not part of a package also seem easy; they should use whatever shebang is appropriate for the system they're written on. It's the `devel` space that's not so clear to me. Sourcing a workspace is like sourcing a python virtual environment, so it almost seems reasonable to make `python` be the version specified by `ROS_PYTHON_VERSION` when a workspace is sourced. However, that might surprise people who automatically source a ROS workspace in their `.bashrc`.

dirk-thomas, 2020-01-03:
The whole problems comes from the combination of two things:
* the broken concept of a devel space
* using the same sources to support a Python 2 distro as well as a Python 3 distro (which makes the shebang lines for one of them incorrect)

Unfortunately I don't think we can get rid of the devel space in ROS 1 - that would break the workflow of too many users (even if it is the right thing to do - that is why it doesn't exist in ROS 2).

I also don't think that we should require different branches (each with their own shebang line). That would add a significant burden on all maintainers.

I do think the error message about non-unique executables ([[1](https://github.com/ros/ros/blob/2b307e1854ebf3954a5b4110f891278f555e79a6/tools/rosbash/rosbash#L248)) is useful for users to catch problems where a package contains multiple executables with the same name. How about punching some holes in that safety net (instead of removing it all together)? In the case there are multiple non-unique executable we should only allow to automatically pick the first one if (and only if) there are exactly two executables (this should be trivial to check) - one in the devel space and one on the source space (not sure if this is possible to check?).

---

A package which wants a "clean" solution should simply branch for Noetic and update the shebang lines to `python3`.

sloretz, 2020-01-03:
> Unfortunately I don't think we can get rid of the devel space in ROS 1 - that would break the workflow of too many users

:(

> I also don't think that we should require different branches (each with their own shebang line). That would add a significant burden on all maintainers.

IIUC that's a reason to eliminate python2or3, but leaves the option of modifying `roslaunch`. It could be argued this eliminates any option using a devel space relay for test scripts since modifying downstream cmake is a similar burden.

> For those scripts [only used for testing but is not being installed,] no CMake is invoked and therefore we can't generate the relay script automatically. Either the package needs to be touched to generate a relay script in the devel space for this or the (launch) test needs to be updated to consider ROS_PYTHON_VERSION. (I don't see a way to satisfy this case without modifying the package itself.)

Seems that way. @mikepurvis's idea of modifying launch looks like it would solve the case for scripts invoked with `rostest` without requiring downstream changes. I'm inclined to move forward with it.

mikepurvis, 2020-01-03:
It's really an issue for any python script invoked from the develspace, not just helper nodes from rostests. We deal with this by just shipping a `<ws>/bin/python` wrapper right in our bundle. This is definitely far too invasive to be generic, but it's what's worked for us through our 2->3 transition.

But yeah, a solution that prepends the `PATH` for any roslaunch subprocess seems reasonable to me, as long as it doesn't prepend it in the `setup.sh`. The only gap then is for scripts that are invoked directly or via `rosrun`, which I'd guess is a tiny minority of them.

If we don't like messing with `PATH`, the other way to achieve this could be something goofy like examining each src executable for a python-ish shebang and if found, auto-prefixing the `<node>` element with the correct executable. Still doesn't fix the situation for non-roslaunched things, but it does actually mean that `/usr/bin/python` shebangs could be caught, which the PATH method would fail for.

sloretz, 2020-02-01:
I updated the issue description with the options I'm aware of. I like the idea of `python` being a special executable in only the devel space as I'm hopeful it's broad enough to cover all cases in the devel space; however, I haven't figured out how to implement it. It seems like it might require changes to all three of `catkin_make`, `catkin_make_isolated`, and `catkin-tools`.

dirk-thomas, 2020-02-06:
> I like the idea of python being a special executable in only the devel space

I am worried that this is too broad. Users might source their workspace in their bashrc and with have override the Python executable for everything in their terminal.

sloretz, 2020-02-06:
> I am worried that this is too broad. Users might source their workspace in their bashrc and with have override the Python executable for everything in their terminal.

I was assuming devel spaces where less likely to be sourced in one's workspace

> This option is proposed to only apply to the devel space, not an install space. I'm assuming devel spaces are less likely to be automatically sourced, so the change to the python executable is less likely to affect non-ROS work.

dirk-thomas, 2020-02-07:
> I was assuming devel spaces are less likely to be sourced in one's .bashrc

I have seen many users source their custom build workspaces - including devel spaces which are the default with `catkin_make(_isolated)` so I wouldn't second that assumption.

sloretz, 2020-03-10:
Fixed by #1879

closed 2020-03-10 by sloretz

sloretz, 2021-11-22:
> I think this change fail to catch the error, where package maintainer forget to use catkin_install_python within their CMakeLists.txt

@k-okada I'm surprised. I think #1830 is supposed to resolve exactly that case. Using `catkin_install_python` on tests has the downside of installing test nodes, meaning they get included in the debian package (Not applicable to `test_rostest` since it isn't released).

This sounds like a bug. Would you be willing to open a new issue?