ros/ros_comm #1832: Core ROS tools use #!/usr/bin/python shebang rather than #!/usr/bin/env python

gstorer, 2019-10-18:
The core ROS tools like roslaunch use #!/usr/bin/python shebang rather than #!/usr/bin/env python .

I'm trying to develop a system using a virtualenv for Python. This system is using a customized logging configuration file that points to a custom formatter. The custom formatter uses third party libraries that are installed in the virtualenv. This works fine for everything except roslaunch which insists on using the system python at /usr/bin/python which doesn't have my third party libraries.  I can work around this but the fact that the source code has /usr/bin/env python in it while my installed binaries have the latter has me concerned. 
https://github.com/ros/ros_comm/blob/melodic-devel/tools/roslaunch/scripts/roslaunch

My guess is that the shebang line is overwritten by the catkin build process (maybe the flag DPYTHON_EXECUTABLE ?). Why is this done? Surely the more universal env line would be better. Also, I could not find exactly where this change takes place. If someone could point me to the relevant build script that would be great.

If this should be on answers.ros.org instead let me know. I thought this was more of a development question/issue though.

mikepurvis, 2019-10-18:
The shebang mutation happens as part of the `catkin_install_python` cmake function:

https://github.com/ros/catkin/blob/3d4fa72a18c7faa061f431d7727c5a87ba1a5b33/cmake/catkin_install_python.cmake#L13

To be honest, I don't know what the original motivation was for this, but it makes it possible to do a lot of nice things such as building a Python installspace that uses `path/to/my/python3` even if the `python` in your `PATH` is still Python 2. It's also pretty critical for building a sane ROS environment on systems like MacOS where you might have multiple possible Python binaries (system, brewed, macports, nix, etc). In my (now unmaintained) build scripts for ROS on Mac, I explicitly pass which python to use, which ensures that CMake scripts configuring bindings and so on aren't left guessing about which interpreter library to link against:

https://github.com/mikepurvis/ros-install-osx/blob/4e342f604960e291320e6acba62ff349747707d5/install#L176-L185

Definitely this can be a source of confusion if you're unwittingly relying on that explicit shebang in an installed package, and then you run the source of it from a develspace and get different behaviour because your env delivered a different interpreter from the one in the cooked shebang.

mikepurvis, 2019-10-18:
Okay, I went digging and it looks like this change was made in 2014 and was largely about supporting Python 2/3 bilingualism:

https://github.com/ros/catkin/pull/574#issuecomment-31712605

gstorer, 2019-10-21:
Ok cool, if its just about managing the 2/3 versions then overriding everything with my virtualenv will be fine. I understand the need for this function in the build scripts but the choice to use it with the Ubuntu pre-built packages still seems odd. Python defaults to version 2 in the supported Ubuntu releases so why make use of this feature? Perhaps it is just a quirk of history?

dirk-thomas, 2020-02-03:
I assume your question was resolved. Most source code uses `#!/usr/bin/env python` (if not please feel free to create a PR to update it). On installation the shebang line is re-written to the concrete Python interpreter being used to install the package (which is standard Python behavior).

Please feel free to continue commenting on the closed ticket.

closed 2020-02-03 by dirk-thomas