ros/ros_comm #1836: bag.py is slow

jomach, 2019-11-17:
Hi all, 
I'm trying to use Bag.py to load 1.4T bag file. Basically it takes  a lot of memory even if I do not read the index (Potentially a bug ?).  Furthermore it takes ages as it only uses one core.

I think we should rewrite Bag.py to have a multi-threaded read based on file chunks.

So basically if we want to compress data on a cluster by using rosbag compress it will take 600m to compress 1.4T even that I do not see any resources bottleneck.

dirk-thomas, 2020-02-03:
Please consider to contribute PRs to fix / improve anything you see.

Since this ticket is just to broad and has no clear acceptance criteria I will go ahead and close it for now.

closed 2020-02-03 by dirk-thomas

witignite, 2021-02-07:
I experienced similar problem using the rosbag's C++ API in Melodic. Specifically, trying to read multiple large bag files using multiple threads doesn't improve the speed, compared to just reading one by one large bag file sequentially; it is even slower. I suspect that the bottleneck is reading bag files from the hard disk; I probably hit the disk access limitation.

@dirk-thomas  I was just wondering if there is a way to set the buffer size to some value large like ~1 Gb so that it reads this large chunk into memory?

dirk-thomas, 2021-02-07:
@witignite Sorry,I am not the maintainer of this repository anymore. Maybe ask your question on answers.ros.org instead.

jomach, 2021-02-09:
@witignite that is the block size of your disk basically. You can see if that helps you. This will not cut it if you have 200GB backfiles