ros/ros_comm #1840: <class 'struct.error'>: 'required argument is not a float' when writing 'data: [3.0, 2.1]'

Algabri, 2019-11-20:
I think that the issue is silly.

 I'd like to run the code on two computers and I need to use `list`. I followed this [http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber(python)](Tutorials). I used my PC as a talker and computer of the robot as a listener.

when running the code on my PC, the output is good as I needed.
```
[INFO] [1574230834.705510]: [3.0, 2.1]
[INFO] [1574230834.805443]: [3.0, 2.1]
```
but once running the code on the computer of the robot, the output is:
```
Traceback (most recent call last):
  File "/home/redhwan/learn.py", line 28, in <module>
    talker()
  File "/home/redhwan/learn.py", line 23, in talker
    pub.publish(position.data)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 886, in publish
    raise ROSSerializationException(str(e))
rospy.exceptions.ROSSerializationException: <class 'struct.error'>: 'required argument is not a float' when writing 'data: [3.0, 2.1]'
```

full code on PC:
```
#!/usr/bin/env python
import rospy
from std_msgs.msg import Float32
x = 3.0
y = 2.1

def talker():
    # if a == None:
    pub = rospy.Publisher('position', Float32, queue_size=10)
    rospy.init_node('talker', anonymous=True)
    # rospy.init_node('talker')
    rate = rospy.Rate(10) # 10hz

    while not rospy.is_shutdown():
        position = Float32()
        a = [x,y]
        # a = x
        position.data = list(a)
        # position.data = a
        # hello_str = [5.0 , 6.1]
        rospy.loginfo(position.data)

        pub.publish(position.data)
        rate.sleep()

if __name__ == '__main__':
    try:
        talker()
    except rospy.ROSInterruptException:
        pass
```
full code on the computer of the robot:

```
#!/usr/bin/env python
import rospy
from std_msgs.msg import Float32

def callback(data):
    # a = list(data)
    a = data.data
    print a

def listener():

    rospy.init_node('listener', anonymous=True)

    rospy.Subscriber("position", Float32, callback)

    # spin() simply keeps python from exiting until this node is stopped
    rospy.spin()

if __name__ == '__main__':
    listener()
```
 when using one number as float everything is OK.

I understand how to publish and subscribe to them separately as the `float` but I'd like to do it as `list`

Any ideas or suggestion, it would be appreciated.

flixr, 2019-11-21:
This would rather be a question for [ROS answers](https://answers.ros.org/questions/)
But what you need to do is use (or create) a message type that holds an array (list) of floats and not only a single value.

jspricke, 2019-11-21:
closing as not an issue. Thanks Felix for answering.

closed 2019-11-21 by jspricke