ros/ros_comm #1845: crash with "error in poll (-1)" after ca 30 min

simonschmeisser, 2019-12-04:
Hi,

I have a very frightening crash here after ca 30 min of letting my ROS based application run idle.

backtrace is:
```
1  __find_specmb                                printf-parse.h   108  0x7f57d6b0ffea 
2  __parse_one_specmb                           printf-parsemb.c 403  0x7f57d6b0ffea 
3  printf_positional                            vfprintf.c       1771 0x7f57d6aebfa9 
4  _IO_vfprintf_internal                        vfprintf.c       1688 0x7f57d6aefcba 
5  ___vsnprintf_chk                             vsnprintf_chk.c  63   0x7f57d6bc5169 
6  XmlRpc::XmlRpcUtil::error(const char *, ...)                       0x7f57d89250d5 
7  XmlRpc::XmlRpcDispatch::work(double)                               0x7f57d891f174 
8  ros::XMLRPCManager::serverThreadFunc()                             0x7f57d8befee8 
9  ??                                                                 0x7f57d7c6dbcd 
10 start_thread                                 pthread_create.c 463  0x7f57d7a446db 
11 clone                                        clone.S          95   0x7f57d6bb488f 
```

The error message it tries to print is: "Error in XmlRpcDispatch::work: error in poll (cDispatch::work: error in poll (-1)."

we are running melodic on ubuntu bionic.

I am at loss for how to debug this :(

jspricke, 2019-12-04:
I guess this it not enough to help you, could you provide a minimal example to reproduce the issue? (At least our systems run well over 30min without problems, so not sure this is an ros_comm issue at all ;) ).

simonschmeisser, 2019-12-05:
(Some of our systems are running in production for a few years now, being rebooted twice a year at most :) )

So I think I tracked this down to a lone ```NodeHandle``` that was created and destroyed before the main system startup. So as far as I understand this would create and register a ros node and then shut it down again. Maybe its network connection gets disconnected/cleaned up after ca 30min? The main node, which gets started later on during the process, then looses it's connection?

I can try to write a MWE

ahoarau, 2020-01-31:
Surely related to https://github.com/ros/ros_comm/pull/1651

dirk-thomas, 2020-02-04:
> I can try to write a MWE

@simonschmeisser Please do so. Until then I will close this ticket. Please comment with the extra information and it can be reopened.

closed 2020-02-04 by dirk-thomas

peterlixue, 2020-03-20:
I have some problem with the author, gdb output the same infomation when my gui app suddenly crashed about 20 minutes.

dirk-thomas, 2020-03-20:
@peterlixue Please don't comment on closed issues if you comment is not directly related to it.