ros/ros_comm #1852: Roscore Error

kvkpraneeth, 2019-12-20:
``praneeth@praneeth-Lenovo-Legion-Y540-15IRH-PG0:~/catkin_ws/src$ roscore
... logging to /home/praneeth/.ros/log/f4804f00-22ff-11ea-bce8-907841bf4730/roslaunch-praneeth-Lenovo-Legion-Y540-15IRH-PG0-5603.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://praneeth-Lenovo-Legion-Y540-15IRH-PG0:40321/
ros_comm version 1.14.3

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES

Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 322, in main
    p.start()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/parent.py", line 288, in start
    self.runner.launch()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/launch.py", line 662, in launch
    self._setup()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/launch.py", line 637, in _setup
    launched = self._launch_master()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/launch.py", line 400, in _launch_master
    validate_master_launch(m, self.is_core, self.is_rostest)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/launch.py", line 83, in validate_master_launch
    if not rosgraph.network.is_local_address(m.get_host()):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosgraph/network.py", line 175, in is_local_address
    local_addresses = ['localhost'] + get_local_addresses()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosgraph/network.py", line 217, in get_local_addresses
    import netifaces
ModuleNotFoundError: No module named 'netifaces'

dirk-thomas, 2020-02-03:
As the error message is stating you are missing the Python module `netifaces`.

closed 2020-02-03 by dirk-thomas

harnix, 2020-03-04:
The error remains despite have netifaces manually installed

dirk-thomas, 2020-03-04:
I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

ranai-srivastav, 2025-01-28:
For anyone who found this page while looking for a solution to this problem, the problem is with your default Python version of installed python. If you recently changed the system default version of Python on your system, this is caused by that. It is safe to say changing the version of the default Python is usually a bad idea, and using `venv` or `conda` is definitely the way to go. 

On Ubuntu 20.04, the default Python version is 3.8 and I think it should stay that way. Try `python --version` or `python3 --version` to see what version of Python you are running and update that with `sudo update-alternatives --config python` and `sudo update-alternatives --config python3`