ros/ros_comm #1857: `rostopic pub <topic_name>` does NOT shows simulation time when /use_sim_time is set to true

heyzude, 2020-01-11:
in ROS, when you set `/use_sim_time` as `true` (by `rosparam set /use_sim_time true`)and do `rostopic pub <topic_name>` , 
You can see wall clock time (system time), not simulation time (ex, starting from the moment when Gazebo started).

I inspected this problem and found that `rostime.py` under `/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/` might be the culprit or deeply related to the problem.

In `rostime.py`, there is a python global variable `_rostime_current`. It is initialized as `None` at the beginning.
At `_set_rostime(t)` `_rostime_current` is assigned as parameter `t`'s value.

And when any outer method or node requires rostime, the `get_rostime()` method returns `_rostime_current`.

However, sometimes I found that even after `_set_rostime(t)` is called `_rostime_current` is still `None.
Furthermore, Even when `/use_sim_time` is true, the parameter `t` of `_set_rostime(t)` is always wall-time.

I tried to fix this, but other files were involved with value of `t`, and those seemed to be very core files...

For now I just made up a "workaround",  which is to insert a few lines in `get_rostime`: Check whether `/use_sim_time` is true and if it is, explicitly get the `/clock` topic and return that value, like below.

    #check /use_sim_time is true or not by getting rosparam value
    use_this_time = rospy.wait_for_message("/clock", Clock)
    return Time(use_this_time.clock.secs, use_this_time.clock.nsecs)

What would be right way to fundamentally fix This bug?

dirk-thomas, 2020-02-03:
`rostopic pub <topic_name>` has nothing to do with `simtime`. The node which publishes the message needs to publish simtime instead of wall time. From your description it is unclear what exact node that is and how you use it. Therefore I will go ahead and close this ticket for now.

Please feel free to continue commenting and provide complete context as well as reproducible steps and the issue can be reopened if necessary.

closed 2020-02-03 by dirk-thomas

heyzude, 2020-07-19:
I don't have the laptop computer I used to use in the company I previously worked for.
So I can't tell how to reproduce and whether the bug was fixed or not now.
Sorry that I forgot some context details related to the bug.
But I know that there was similar issue at ros rust: https://github.com/adnanademovic/rosrust/issues/36
And the snapshots below are confluence wiki pages that I wrote in the past, regarding this error and the workaround to deal with this error

Issue:
[image]
Workaround:
[image]
For people who are working on ros python and willing to dig down the issue I would really appreciate.