ros/ros_comm #1861: Integrating with OSS-Fuzz

Google-Autofuzz, 2020-01-17:
Greetings ros_comm developers and contributors,

We’re reaching out because your project is an important part of the open source ecosystem, and we’d like to invite you to integrate with our [fuzzing](https://www.owasp.org/index.php/Fuzzing) service, [OSS-Fuzz]( https://opensource.googleblog.com/2016/12/announcing-oss-fuzz-continuous-fuzzing.html ). OSS-Fuzz is a free fuzzing infrastructure you can use to identify security vulnerabilities and stability bugs in your project. OSS-Fuzz will:

- Continuously run at scale all the fuzzers you write.
- Alert you when it finds issues.
- Automatically close issues after they’ve been fixed by a commit.

Many widely used [open source projects]( https://github.com/google/oss-fuzz/tree/master/projects ) like OpenSSL, FFmpeg, LibreOffice, and ImageMagick are fuzzing via OSS-Fuzz, which helps them find and remediate [critical issues]( https://bugs.chromium.org/p/oss-fuzz/issues/list?can=1&q=status%3AFixed%2CVerified+Type%3ABug%2CBug-Security+-component%3AInfra+ ). 

Even though typical integrations can be done in < 100 LoC, we have a [reward program]( https://www.google.com/about/appsecurity/patch-rewards/ ) in place which aims to recognize folks who are not just contributing to open source, but are also working hard to make it more secure.

We want to stress that anyone who meets the eligibility criteria and integrates a project with OSS-Fuzz is eligible for a reward.

If you're not interested in integrating with OSS-Fuzz, it would be helpful for us to understand why—lack of interest, lack of time, or something else—so we can better support projects like yours in the future.

If we’ve missed your question in our [FAQ]( https://google.github.io/oss-fuzz/faq/ ), feel free to reply or reach out to us at oss-fuzz-outreach@googlegroups.com.

Thanks!

Tommy
OSS-Fuzz Team

tfoote, 2020-01-22:
Having Fuzz tests would be valuable, though the limitation of a single process is going to limit the value of the testing for ROS overall as we're designed to work in large distributed networks.

With our federated development model and large number of community packages I think that focusing on creating a ROS fuzzing package would have the highest impact instead of picking one repository like this one.

I'm not familiar with the fuzzing setup but if a CMake macro could be implemented to add fuzz tests like we add gtests with catkin_add_gtests http://wiki.ros.org/gtest

However for the highest impact I'd suggest looking at adding this to our next generation of ROS. I suspect that an ament_lint implementation would be the most powerful. 

We have designed the ability to have many linters and a fuzzing linter could be relatively easily added. 
Design: https://design.ros2.org/articles/ament.html
Standard linters: https://github.com/ament/ament_lint
A guide to making a new linter: https://ubuntu.com/blog/how-to-add-a-linter-to-ros-2

These linters can then be easily deployed by all developers in the ROS 2 ecosystem. 

I don't think that this is something that we're going to actively pursue but would welcome contributions in this direction.

tfoote, 2020-01-28:
Since this doesn't have anything specific to track to do I'm going to close it out. We'd be happy for contributions in this direction though.

closed 2020-01-28 by tfoote