ros/ros_comm #1862: Start/end date in rosbag is always Dec 31 1969

pauljurczak, 2020-01-20:
Here is what I'm getting for a freshly recorded rosbag:
```
paul@paul-desktop:~/st/subt/bin$ rosbag info a.bag
path:        a.bag
version:     2.0
duration:    14:44s (884s)
start:       Dec 31 1969 17:00:52.61 (52.61)
end:         Dec 31 1969 17:15:37.46 (937.46)
size:        30.4 GB
messages:    17698
compression: none [17698/17698 chunks]
types:       sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181]
topics:      X1/rgbd_camera/depth/points   17698 msgs    : sensor_msgs/PointCloud2
```

mikepurvis, 2020-01-20:
Is this in an environment where `use_sim_time` was set to true?

Was it a VM, docker, chroot, snap, or some other exotic scenario?

pauljurczak, 2020-01-20:
`use_sim_time` was set to `true` by Ignition simulation. Not a VM, docker, chroot, snap, or some other exotic scenario. Does it mean the date is as expected? If so, clock time matching the real world (`17:00:52.61`), but not the date is surprising.

I will go to ROS Answers first, the next time.

mikepurvis, 2020-01-20:
If `use_sim_time` is set, then you also need to be publishing time to the `/clock` topic. I'd expect that Gazebo/Ignition would handle this on its own, however I'm not familiar with the newer architecture there. Either way, I'd consult the docs on integrating your version of Gazebo with ROS.

closed 2020-01-20 by mikepurvis

tfoote, 2020-01-23:
This is expected as the simulator starts counting from time from zero. A representation of that in human timestamps will be January 1st 1970. And if you're in a timezone behind GMT you can get into Dec 31st 1969. If you ran your bag recorder for long enough it would wrap into 1970. 

https://en.wikipedia.org/wiki/Unix_time