ros/ros_comm #1863: Forward declaration of struct in network.cpp on FreeBSD

dodsonmg, 2020-01-22:
`network.cpp` in `roscpp` includes `ifaddrs.h`.

On Linux, `ifaddrs.h` includes `sys/socket.h`, which defines `struct sockaddr`.

On FreeBSD, `ifaddrs.h` does *not* include `sys/socket.h`, though `sys/socket.h` still defines `struct sockaddr`.

When I try to build on FreeBSD, I get a series of `forward declaration of struct` errors when building `network.cpp` associated with pointers to `sockaddr` structures in the code block from lines 126 to 176.

The problem is fixed by including `sys/socket.h`, but presumably this isn't desirable, as the target OS for ROS is Ubuntu, and this include statement would be superfluous.

Ideally, I'd like to modify the include block of `network.cpp` as follows:

```
#ifdef HAVE_IFADDRS_H
    #include <ifaddrs.h>
    #include <sys/socket.h>  // added
#endif
```

How can I do this smarter, to support an acceptable PR?

mikepurvis, 2020-01-22:
The proposed modification looks fine to me. Just include a comment noting to future maintainers that it's there because FreeBSD needs it, so someone doesn't mistakenly "clean it up" down the road.

We have plenty of other similar hacks to ease building on non-Ubuntu systems, especially Windows and macOS.

dodsonmg, 2020-01-27:
Closing, as this is now tracked by PR #1864.

closed 2020-01-27 by dodsonmg