ros/ros_comm #1880: Roscpp won't compile on Arch Linux because it expects Boost 1.61 or above and fails to recgonize my Boost 1.72.0 version.

djohn49, 2020-02-09:
I am currently trying to build Ros on Arch Linux through yay/aur as documented [here](wiki.ros.org/melodic/Installation/ArchLinux). When yay gets to roscpp, however, the build fails with the following error (happens about a half hour after starting the command to install the full desktop package set):
```
-- Found Boost: /usr/lib64/cmake/Boost-1.72.0/BoostConfig.cmake (found version "1.72.0") found components: chrono filesystem system 
CMake Error at CMakeLists.txt:30 (message):
  roscpp requires Boost 1.61 or above.

-- Configuring incomplete, errors occurred!
See also "/home/david/.cache/yay/ros-melodic-roscpp-git/src/build/CMakeFiles/CMakeOutput.log".
See also "/home/david/.cache/yay/ros-melodic-roscpp-git/src/build/CMakeFiles/CMakeError.log".
==> ERROR: A failure occurred in build().
    Aborting...
error: target not found: ros-melodic-roslint
error: target not found: ros-melodic-urdf-parser-plugin
error: target not found: ignition-cmake
error: target not found: ignition-tools
```
It looks like whatever version check is being used can't work correctly with Boost 1.72.0 for some reason.

dirk-thomas, 2020-02-10:
You might want to print the Boost version as reported by CMake by adding the following line above the line resulting in the fatal error:

```message("Boost_VERSION ${Boost_VERSION}")```

john01dav, 2020-02-11:
Isn't that what's already shown in the first line of my log? It says that the version is 1.72.0.

dirk-thomas, 2020-02-13:
I will close the ticket for now. Please feel free to comment on the closed ticket with the requested information and if necessary the ticket can be reopened.

closed 2020-02-13 by dirk-thomas

tspicer01, 2020-02-27:
@dirk-thomas   We just upgrade boost (1.71) and hit the same issue.
It looks like there is a new policy on how the boost version is defined
https://cmake.org/cmake/help/v3.15/policy/CMP0093.html?highlight=cmp0093

I've changed my local CMakeList.txt to version check on Boost_VERSION_MACRO instead of Boost_VERSION and that is consistent with the old behaviour.

```
Boost Version found 1.71.0
Boost Version Macro found 107100
```